MiczFlor/RPi-Jukebox-RFID #501: Rpi 1 model B - Not USB RC522 - GPIO buttons not working

dvalob, 2019-02-26:
Hi!

I am having troubles to use the GPIO buttons. I am using RPI 1 model B and a GPIO RFID reader. The reader is working fine and the gpio service is enabled and I see it with htop every 2/3 seconds. I also tried as said in #383 to put the pull_up to false. I also tried to comment out the non used commands and paid attention to GPIO/pin number mistake (#487). The GPIO numbers given in the script are not avail in my PI model, so I needed to change them.

I put the arcade buttons that you recommend as follows:
- Prev: pins 12 (GPIO18) - 14 (GND)
- Next: pins 9 (GND) - 11 (GPIO17)
- Pause: pins 25 (GND) -26 (GPIO7)

GPIO and RIFD are enabled on web interface.

Any clue??

I guess that GPIO are working because RIFD reader is on it. But I have no more clue for the moment.

dvalob, 2019-02-26:
I used a small script to see what is being read o this GPIO:

[image]

And it happens that the Button press is seen, then when I release it, I see a toggling between pressed/not pressed. 

[image]
I tried the same using a wire to simulate a contact/not contact and I have the same result. So it does not seem to be linked to the HW. Maybe the pull up resistor would solve this toggling value?

I suppose that the Rpi does not like to much to see this value toggling in order to go to next/previous track, etc.

Can anyone having these buttons working can run the given script and tell me if you have the same output?? @bofelix? @georgbachmann?

Ardios, 2019-02-28:
@dvalob 

how many Pins have your Acarde Button? 2 or 3?

dvalob, 2019-02-28:
I have 3 buttons, each one connected to 2 pins (one GPIO and one GND). I am not using a "common ground", but 3 different GND pins

dvalob, 2019-03-04:
Hi! I finally had it working after several resets. The only new thing that I did is to comment all the unsused lines in the gpio-buttons.py (I had forgot lines  between 62 and 75). I do not use external pull-up resistors.

I also added a rotary switch for volume and I used its pushbutton (SW pin) to properly shutdown the PI.

At the end, my pin allocation is :

[image]

Daniel

closed 2019-03-04 by dvalob