MiczFlor/RPi-Jukebox-RFID #502: How to delete folders/Playlists in web Gui

Alwiz79, 2019-03-01:
I have some folders/playlists that I don't use anymore.
I managed to use the cards for another folder/playlist, but the old folder is still visible in the web gui. There no card linked to this folder anymore.
How can I remove this?

MiczFlor, 2019-03-01:
Hi @Alwiz79 
that's a good question - and came up in another thread. I will implement this in the webUI soon. For now, please connect to the Phoniebox via SAMBA or SSH as described here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#connecting-to-the-phoniebox-to-add-files

closed 2019-03-01 by MiczFlor

Alwiz79, 2019-03-01:
@MiczFlor not sure how this helps.
Starting playlists via webgui is not the problem.
I just want delete the unwanted playlists. is this possbible in some file (via ssh) ?

andyyo82, 2020-06-24:
Is there any solution for this Problem? I have the Same Problem and want to delete unused Playlists/Folders.

Thank You

SirY4, 2020-07-08:
I tried to reopend this thread as a new issue. I am hoping for some feed back.