MiczFlor/RPi-Jukebox-RFID #504: List of working web radios

dvalob, 2019-03-04:
Hi

I propose to create a list or something similar to share webradios urls that are working on ours phonieboxes. 

I use the following ones:

Spanish:
- los40: http://20043.live.streamtheworld.com/LOS40.mp3
- Kiff FM: http://kissfm.kissfmradio.cires21.com/kissfm.mp3
- cadena 100: http://195.10.10.222/cope/cadena100.mp3
- Babyradio (Kids): http://sd396babyradio1.lcinternet.com:8000/stream

French:
- Radio Barbouille (Kids): http://sd396babyradio1.lcinternet.com:8000/stream

MiczFlor, 2019-05-25:
Hi @dvalob 
Possibly this might be a nice addition to the wiki? The issue thread doesn't seem to be the right place. What do you think?

dvalob, 2019-05-25:
Sure! as you wish :)
I was just thinking to somewhere where everybody can easily participate :)