MiczFlor/RPi-Jukebox-RFID #505: RFID-Reader: eigene Installation (kein USB)

tugendbund, 2019-03-04:
Hallo zusammen,

ich bin gerade dabei, mich in das Projekt einzuarbeiten und bin in dieser Welt ziemlich neu.

Auch ich habe das Problem mit dem USB-Reader von Neuftech und arbeite mich diesbezüglich durch die Forumsbeiträge.

Da ich damit überhaupt nicht weiterkomme, habe ich mir selbst einen Reader mit Chip, Breadboard und Jumperkabel nach [dieser Anleitung](https://pimylifeup.com/raspberry-pi-rfid-rc522/) gebaut. 
Reader und Writer funktionieren über die Konsole einwandfrei. 

Zurück auf der Phoniebox-Oberfläche passiert etwas Seltsames: In dem Formular-Feld für die CardIDs wird zwar etwas gescannt (anders als beim USB-Reader) und dem Musik-Ordner zugeordnet, aber anstatt der CardIDs werden die Befehle geschrieben, die ich im Terminal eingegeben habe. Zum Beispiel "Write.py" oder ls. Ich habe in die Datei Latest-RFID unter Settings geschaut und sehe, dass statt der ID tatsächlich "ls" steht.

Hat jemand eine Idee, was da schief läuft?

Ardios, 2019-03-06:
@tugendbund 

Wenn du nach deiner Verlinkten Anleitung vorgegangen bist, müsstest du 7 Kabel vom RC-522 angeschlossen haben. Für dieses Projekt musst du aber alle 8 Kabel des RC-522 verbinden.

Hier ein Link zur Wiki inkl. Anschlussbildern, das braune Kabel (IRQ) müsste noch angeschlossen werden:
[](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Wiring_for_RC522_card_reader)

Ansonsten hier noch ein Wiki Eintrag für die Installation:
[](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/RFID-Reader-Special)

Alternativ hier noch ein Link zur einer Anleitung von mir, wenn du dich daran hälst sollte es eigentlich laufen:
[](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/11#issuecomment-456071515)

Grüße Ardios

tugendbund, 2019-03-06:
Vielen Dank für deine Antwort.

Habe 8. Kabel installiert und alle Anweisungen befolgt: wieder erfolglos.

In den Settings ist der Reader Enabled und in der Datei DeviceName.txt steht "MFRC522".

Habe in  Reader.py.experimental bzw. nachher dann ja Reader.py den Kommentar vor import MFRC522
weggemacht. Keine Änderung.

Wie kann ich den Reader denn über die Konsole ausprobieren? Wie gesagt, der gleiche Reader nach besagter Anleitung hat zumindest im Terminal funktioniert.

Ardios, 2019-03-06:
@tugendbund 

hm das ist komisch...ich nehme gleich nochmal eine Neuinstallation vor und teste es dann auch nochmal.

Um zu testen ob der Reader mit allen 8 Kabeln funktioniert, klone bitte folgendes Paket:

`git clone https://github.com/ondryaso/pi-rc522.git`

und dann das example script einmal ausführen mit:

`python /home/pi/pi-rc522/examples/Read.py`

Was bekommste da für einen Output?

Grüße

tugendbund, 2019-03-06:
> /home/pi/.local/lib/python2.7/site-packages/pirc522/rfid.py:51: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
>   GPIO.setup(pin_rst, GPIO.OUT)
> Starting

Das Problem hatte ich bei der anderen Installation auch und habe im dortigen Reader 

> GPIO.setwarnings(False) 

ergänzt. Aber das macht hier ja keinen Sinn...

Danke und Grüße

Ardios, 2019-03-06:
@tugendbund

Die Fehlermeldungen sind soweit normal...kommt noch was nachdem "Starring" wenn du eine Karte vor den Reader hälst? Wenn nicht, bitte nochmals Kabel überprüfen.

Und könntest du schauen ob spidev und pi-rc522 richtig installiert wurden? 

pip install spidev und pip install pi-rc522

Grüße

tugendbund, 2019-03-06:
> pip install spidev und pip install pi-rc522
> Collecting spidev
> Collecting und
>   Could not find a version that satisfies the requirement und (from versions: )
> No matching distribution found for und
> 

Komisch, ich bin mir ziemlich sicher, dass das bei der Installation vorhin geklappt hat. 

Nach "Starting" kommt nichts mehr. 

Kabel sind überprüft. Nur eine kleine Frage: Habe "Hardware with fixed pin assignment" verwendet. Richtig,oder? Da steht bei GPIO24 Pin 18 statt 8. Ich habe angenommen, dass das einfach ein Tipp-Fehler ist und hier ja sowieso auch andere Pins gewählt werden können.

Ardios, 2019-03-06:
@tugendbund 

Sorry, hatte es gerade schnell mit dem Handy geschrieben.

Also die Befehle müssen einzeln ausgeführt werden und ich habe das sudo vergessen:

`sudo pip install spidev`

und danach dann

`sudo pip install pi-rc522`

Hier nen Link zur Anschlussbelegung, ich habe mich immer daran gehalten also habe nie andere Pins probiert...mein rc-522 habe ich so angeschlossen wie auf den Bildern in der Wiki zu sehen ist.

https://github.com/ondryaso/pi-rc522

Einfach ein bisschen runterscrollen bis du "Connecting" siehst

Grüße
Ardios

tugendbund, 2019-03-06:
Oh Mann, das mit dem UND hätte ich auch selbst sehen können! Sorry!

Also, alles nochmals überprüft. Nach "starting" kommt nichts mehr.

Ich habe mir auf die Pins eine [Schablone](https://www.computerhilfen.de/info/wp-content/uploads/2016/04/computerhilfen_de-gpio.png) gelegt, deren Nummerierung nicht ganz mit deiner übereinstimmt (GPIO24 auf Pin 8 statt 18). Spielt das eine Rolle?

Ardios, 2019-03-06:
@tugendbund 
Was hast du denn für einen Raspberry? Normalerweise sollten die Pins schon gleich sein ;)

Ich nutze immer folgende Seite:
https://de.pinout.xyz/#

Grüße
Ardios

tugendbund, 2019-03-06:
Ok, mein Denkfehler. 24->GPIO8.

Also, ist alles korrekt angeschlossen und mehrfach überprüft. Ich habe übrigens einen 3+.

Ardios, 2019-03-06:
@tugendbund 
Echt eigenartig, habe meine Installation gerade auch abgeschlossen aber bei mir klappt es direkt :(

Ich weiß das es nervt, ich habe hier nochmal eine bebilderte Anleitung, könntest du nochmal Punkt für Punkt abgleichen? Ansonsten mal zwischendurch nen Reboot falls noch nicht geschehen

Link:
https://raspi.mobi/viewtopic.php?f=10&t=4

Grüße
Ardios

tugendbund, 2019-03-06:
Also, alles nochmal Schritt für Schritt durchgegangen.

Und: Es funktioniert, ABER mit dem USB-Reader, nicht mit dem selbstgebastelten. Da habe ich das gleiche Problem wie vorher: Nach "starting" passiert nichts mehr...

Mir fällt nichts anderes mehr ein, als meine Karte zu formatieren und das ganze System nochmal ganz neu aufzusetzen. Ich habe jetzt vier Tage lang daran herumprobiert. Wer weiß, woran es liegt.

Ich danke dir trotzdem sehr. Die Anleitung ist total hilfreich und mir ihr funktioniert es wenigstens mit dem USB-Reader.

Du hast mich echt vor dem Wahnsinn gerettet.

Ardios, 2019-03-06:
@tugendbund 
Immer wieder gern :)

Ja wenn du deine Karte nochmal komplett neu aufsetzt, kannste es ja nochmal mit der Anleitung probieren...da du ja schon einen Output über das MFRC Package hattest und du nur den IRQ Pin nachgerüstet hast, sollte es dann ja eigentlich funzen.

Wenn es dann immer noch nicht läuft, dann müsstest du evtl. mal mit einem Multimeter die Verbindungen messen aber ich würde es auch erstmal mit Neu aufsetzen probieren.

Grüße
Ardios

MiczFlor, 2019-03-25:
Hi @Ardios @tugendbund 
wenn das eher Hardware / Lötstellen Probleme sind, können wir dann den thread schließen?

Ardios, 2019-04-06:
@MiczFlor

Wenn @tugendbund keine weiteren Probleme hat, wurde es wohl durch die Neuinstallation behoben. So oder so scheint es sich hier nicht um ein Software Problem zu handeln.

Daher könnte es von meiner Seite her geclosed werden.

Grüße
Ardios

tugendbund, 2019-04-06:
Von meiner Seite aus ok den Thread zu schließen.
Danke.

closed 2019-04-07 by MiczFlor