MiczFlor/RPi-Jukebox-RFID #506: GPIO Buttons Flexible Pinout doesn't work

Josephraclure42, 2019-03-07:
Hi,
Thanks for that amazing project ! 

In order to plug a Hifiberry miniamp, I have to change the GPIO buttons assignement in /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py . 

I checked the "Hardware Pinout overview" page and here is my modifications : 

#shut = Button(3, hold_time=2)
#vol0 = Button(13,pull_up=True)
volU = Button(5,pull_up=True,hold_time=0.3,hold_repeat=True)
volD = Button(6,pull_up=True,hold_time=0.3,hold_repeat=True)
next = Button(23,pull_up=True)
prev = Button(22,pull_up=True)
halt = Button(27,pull_up=True)
#reco = Button(6, pull_up=True) # Choose GPIO to fit your hardware
#play = Button(12,pull_up=True) # Choose GPIO to fit your hardware

But it doesn't work ... I tried to:
sudo systemctl daemon-reload
sudo systemctl enable phoniebox-gpio-buttons
sudo systemctl start phoniebox-gpio-buttons

but no response when I press the buttons . In fact, if I replug the play button in GPIO21 like before ... it works ! 

What am I doing wrong please ? 

Thank you !

Josephraclure42, 2019-03-10:
A clue anyone? I don't understand why it doesn't work ... if I run the script gpio-buttons.py it seems ok , but nothing happen when I push my buttons with new pinout ... 
Many thanks

martinclausen2, 2019-03-23:
@Josephraclure42 could you check what services are running on your box?
sudo systemctl status
Maybe you have several instances running?

MiczFlor, 2019-05-25:
Hi @Josephraclure42 
is this still an issue? If you solved it, please add your solution here and close the thread. Thanks!
All the best, micz

lj2000, 2019-06-09:
Hi @Josephraclure42 

I have a hifiberry miniamp in my setup, too.

And I changed the pinning similar to yours.

Have you thought that the GPIO numbers differ to the hardware pin numbers at the raspberry?

Here are some examples:
**GPIO** <> **HW-Pin**
23 <> 16
24 <> 18
25 <> 22

For details see https://www.raspberrypi.org/documentation/usage/gpio/

splitti, 2019-06-09:
Do you have the OLED Display installed with Option 2?

s-martin, 2020-04-12:
Closing now. @Josephraclure42, if this issue still persist please feel free to reopen the issue.

closed 2020-04-12 by s-martin