MiczFlor/RPi-Jukebox-RFID #510: Add a simple file manager to WebGUI

Ardios, 2019-03-10:
Hello everyone,

today I will show you how to add a file manager to the WebGUI.
There are several ways to include a file Manager, i have chosen iframe.

1.) we need a File Manager PHP Script, i took [Tiny File Manager](https://github.com/prasathmani/tinyfilemanager) from Github:

[](https://github.com/prasathmani/tinyfilemanager)
You need `tinyfilemanager.php` and for other languages you also need `translation.json`. 

2.) Edit `tinyfilemanager.php` Change Line 46 from `$root_path = $_SERVER['DOCUMENT_ROOT'];` to `$root_path = '/home/pi/RPi-Jukebox-RFID/shared';` and Line 50 from `$root_url = '';` to `$root_url = '/home/pi/RPi-Jukebox-RFID/shared';` and save.

3.) Copy `tinyfilemanager.php` and `translation.json` to `/home/pi/RPi-Jukebox-RFID/htdocs/`

Now we have to decide if we want to include the File Manager in `Folders & Files` (manageFilesFolders.php) or if we prefer it as a own page (e.g. filemanager.php).

 **`Folders & Files` (manageFilesFolders.php):**

1.) Edit `manageFilesFolders.php` in `/home/pi/RPi-Jukebox-RFID/htdocs/manageFilesFolders.php` and add on Line 400 after:
```
		<?php
		if ($edition == "plusSpotify") {
		print "
		<legend><i class='mdi mdi-autorenew'></i> ".$lang['manageFilesFoldersRenewDB']."</legend>
		<h4>".$lang['manageFilesFoldersRenewDBinfo']."</h4>
		<a href='".$_SERVER['PHP_SELF']."?scan=true' class='btn btn-success'> ".$lang['manageFilesFoldersLocalScan']."</a>";
		}
		?>
```

this:

```
<fieldset> 
<legend>File Manager</legend>
		
<div style="text-align:center;">
<iframe height="600" width="100%" src="tinyfilemanager.php" align="middle" frameBorder="0"></iframe>
</div>
</fieldset>
```
Save and Done.

**Prefer it as a own page (e.g. filemanager.php):**

1.) Edit inc.navigation.php in `/home/pi/RPi-Jukebox-RFID/htdocs/inc.navigation.php` and add on Line 29 after:

`<li><a href="cardRegisterNew.php" class="mainMenu"><i class='mdi mdi-cards-outline'></i> <?php print $lang['globalRegisterCardShort']; ?></a></li>`

add this:

`<li><a href='filemanager.php' class='mainMenu'><i class='mdi mdi-folder-plus'></i> File Manager</a></li>`

2.) I attached a new File `filemanager.php` only copy to `/home/pi/RPi-Jukebox-RFID/htdocs/` i had edited the `cardRegisterNew.php` for `filemanager.php`

Done.

Please excuse my broken English

Ardios

MiczFlor, 2019-03-10:
Hi @Ardios 
I like the idea. Personally, I do not use the web UI for file management. 
It would be great if you could find a solution using a MIT license, like Phoniebox. Would you know one?
And if I hack this myself: what tasks need to be covered?

Ardios, 2019-03-10:
@MiczFlor 

like this?

[https://github.com/alexantr/filemanager](https://github.com/alexantr/filemanager)

ckuetbach, 2019-07-12:
Just wanted to mention [minio](https://github.com/minio/minio). 

It is a single go-application without any dependencies, which has a file server mode.

There is a download for ARM as well. I think it should not be included into phoniebox, because it was made for different purpose. But if one simply wants to have a web filemanager, this could be sufficient.

```
pi@raspberrypi:~ $ export MINIO_ACCESS_KEY=minio
pi@raspberrypi:~ $ export MINIO_SECRET_KEY=miniostorage
pi@raspberrypi:~ $ ./minio server .
```

The arm (32 bit) version can be downloaded here:
https://dl.min.io/server/minio/release/linux-arm/

There two more arm versions (arm64 and arm6vl).