MiczFlor/RPi-Jukebox-RFID #511: volume set in alsa is ignored

fourstone, 2019-03-11:
Hello,

I am using the following setup:
- Raspberry Pi Zero W
- Adafruit Speaker Bonnet
- RFID Reader connected to extended PINs as described in wiring sheet.

Current Status:
- RFID works with the experimental script 
- sound also works, but with 100% volume always now.

I had problems with the Softvolume not working reliably, i.e. when booting I could not change the volume, then needed to do some steps, reboot and only then it worked.
Now I am trying the version without Soft-Volume and it works quite well - except for Phoniebox which still plays at 100%.
Details: https://learn.adafruit.com/adafruit-speaker-bonnet-for-raspberry-pi/i2s-tweaks
- Step 1 is what I currently have configured. 
- Step2 is with Softvolume and this is also included in the default installation

using mpg123 and playing an mp3 file this comes out at the correct volume and the volume level is also displayed on the Phoniebox GUI.
I tried setting the iFace name to Master and PCM, both behave the same
in MPD I commented out the name, set it to PCM and Master and all behaved the same

The startup sound by the way uses the expected volume, i.e. lesser volume, not 100%
speaker test also works at desired volume

Probably only something small is missing, but I ahve no idea what it is.

thanks, Martin

Output of Analytics Script(exchanged the # with - to make it more readable):
`
************************************
*** PHONIEBOX INFO
*** version: 1.1.9
*** edition: classic
*** Audio_iFace_Name: PCM
*** Audio_Folders_Path: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** Audio_Volume_Change_Step: 3
*** Max_Volume_Limit: 80
*** Idle_Time_Before_Shutdown: 60
*** Second_Swipe: NOAUDIOPLAY
*** Playlists_Folders_Path: /home/pi/RPi-Jukebox-RFID/playlists
*** ShowCover: ON
************************************
*** CONF FILES DEFAULT

*** /etc/samba/smb.conf
-rw-r--r-- 1 root root 9.5K Feb 27 12:05 /etc/samba/smb.conf
   path=/home/pi/RPi-Jukebox-RFID/shared
   path=/home/pi/RPi-Jukebox-RFID/shared/audiofolders
*** /etc/lighttpd/lighttpd.conf
-rw-r--r-- 1 root root 1.1K Feb 27 12:05 /etc/lighttpd/lighttpd.conf
*** /etc/lighttpd/conf-available/15-fastcgi-php.conf
-rw-r--r-- 1 root root 398 Feb 27 12:05 /etc/lighttpd/conf-available/15-fastcgi-php.conf
*** /etc/php/7.0/fpm/php.ini
-rw-r--r-- 1 root root 70K Feb 27 12:05 /etc/php/7.0/fpm/php.ini
*** /etc/sudoers
-r--r----- 1 root root 784 Feb 27 12:05 /etc/sudoers
*** /etc/systemd/system/phoniebox*
-rw-r--r-- 1 root root 296 Feb 27 12:05 /etc/systemd/system/phoniebox-rfid-reader.service
-rw-r--r-- 1 root root 340 Feb 27 12:05 /etc/systemd/system/phoniebox-startup-sound.service
-rw-r--r-- 1 root root 289 Feb 27 12:05 /etc/systemd/system/phoniebox-gpio-buttons.service
-rw-r--r-- 1 root root 291 Feb 27 12:05 /etc/systemd/system/phoniebox-idle-watchdog.service
*** /etc/mpd.conf
-rw-r----- 1 mpd audio 14K Mar 11 11:10 /etc/mpd.conf

music_directory         "/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
- music_directory are changed.
- symbolic links outside of the configured music_directory.
- symbolic links inside of the configured music_directory.
        mixer_control   "PCM"           # optional
-       mixer_control   "PCM"           # optional
*** /etc/dhcpcd.conf
-rw-rw-r-- 1 root netdev 1.7K Sep 10 09:31 /etc/dhcpcd.conf
#static ip_address=192.168.0.10/24
#static ip_address=192.168.1.23/24
#static routers=192.168.0.1
#static routers=192.168.1.1
option domain_name_servers, domain_name, domain_search, host_name
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1
#static domain_name_servers=192.168.1.1
*** /etc/wpa_supplicant/wpa_supplicant.conf
-rw-r--r-- 1 root root 235 Mar 11 08:59 /etc/wpa_supplicant/wpa_supplicant.conf
country=DE
************************************
*** +Spotify Edition

*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Feb 27 13:24 /etc/locale.gen
*** /etc/locale.gen
-rw-r--r-- 1 root root 8.9K Feb 27 13:24 /etc/locale.gen
*** /etc/mopidy/mopidy.conf
ls: cannot access '/etc/mopidy/mopidy.conf': No such file or directory
cat: /etc/mopidy/mopidy.conf: No such file or directory
*** ~/.config/mopidy/mopidy.conf
ls: cannot access '/home/pi/.config/mopidy/mopidy.conf': No such file or directory
cat: /home/pi/.config/mopidy/mopidy.conf: No such file or directory
`

Ardios, 2019-03-11:
@fourstone 

Did you try softwaremixer in the mpd.conf under Audio_output?

Edit `/etc/mpd.conf`:

Look for:

`#       mixer_type      "hardware"      # optional`

and Change it to:

`       mixer_type      "software"      # optional`

Then reboot and try again

Ardios

fourstone, 2019-03-11:
not sure if that is going in the right direction, but will try anyway. MPD already shows the correct volume but plays at 100%.

Ardios, 2019-03-11:
@fourstone 

ah sorry, which device is displayed below alsamixer?

e.g <Master> or <SoftMaster> … etc

Ardios

fourstone, 2019-03-11:
@Ardios 
The problem seems to be solved after a fresh reinstall, it seems it was just some inconsistency. I just reinstalled, used the script setup for the sound, did the rebooting as suggested, installed phoniebox anew, disabled gpio buttons, rebooted, put the experimental script in place and suddenly everything worked.

What is quite funny is that before reinstall I had 2 mixers (Master AND PCM).
now I only have one called PCM. No idea where the difference came from.

Thanks for trying to help :D

For anybody having the same hardware setup, I think maybe I had some of this different last time, but I am not sure what, posting the working configuration:

asound.conf:
```
pcm.speakerbonnet {
   type hw card 0
}

pcm.dmixer {
   type dmix
   ipc_key 1024
   ipc_perm 0666
   slave {
     pcm "speakerbonnet"
     period_time 0
     period_size 1024
     buffer_size 8192
     rate 44100
     channels 2
   }
}

ctl.dmixer {
    type hw card 0
}

pcm.softvol {
    type softvol
    slave.pcm "dmixer"
    control.name "PCM"
    control.card 0
}

ctl.softvol {
    type hw card 0
}

pcm.!default {
    type             plug
    slave.pcm       "softvol"
}
```

activated options in config.txt:

```
dtparam=spi=on

...

# Audio Bonnet
dtoverlay=hifiberry-dac
dtoverlay=i2s-mmap

# disable BT
dtoverlay=pi3-disable-bt

# RFID RC522
device_tree_param=spi=on
dtoverlay=spi-bcm2708
```

closed 2019-03-11 by fourstone