MiczFlor/RPi-Jukebox-RFID #512: Play on chromecast or google home

atwc, 2019-03-13:
Is it possible to launch spotify on other device from Phoniebox, like on the spotify app, to play on chromecast(audio) or google home?

MiczFlor, 2019-03-15:
This does not answer your question but might provide a new train of thought:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Bluetooth

atwc, 2019-03-15:
It could be an option, but be compatible with chromecast or google assistant would allow to have only one Phoniebox for multiple speaker.
With a selector button we could choose to play on the wanted room on a chromecastAudio, a google home or other connected speaker (sonos).

MiczFlor, 2019-03-25:
HI @atwc 
trying to merge threads, I will close this and please jump into this discussion.
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/499
All the best, micz

closed 2019-03-25 by MiczFlor