MiczFlor/RPi-Jukebox-RFID #513: DAC problem (again)

iwishitwassummer, 2019-03-16:
Hi all,

I'm becoming desperate. I can't get the Justboom DAC Zero to work on the Pi Zero W :(
Could someone help?
I bought a header pre-soldered Pi zero W and the DAC is added to it (no stacking header at the moment).
To me it looks like the card is recognized but I can't get any sound. Neither with speaker-test not the phoniebox web.
A headphone is plugged into the headphone jack of the DAC.
Any idea is highly welcome. 
Thanks

Relevant configs:

##########################
--cat /boot/config.txt:
 
dtparam=audio=off
dtoverlay=justboom-dac
 
##########################
--/proc/asound/modules:
 0 snd_soc_justboom_dac
 
##########################
--aplay list:
 
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpijustboomd [snd_rpi_justboom_dac], device 0: JustBoom DAC HiFi pcm512x-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
 
##########################
--cat /proc/asound/cards:
 
 0 [sndrpijustboomd]: JustBoomDac - snd_rpi_justboom_dac
                      snd_rpi_justboom_dac
 
##########################
--mpc outputs:
 
Output 1 (JustBoom) is enabled
 
##########################
--cat /etc/asound.conf:
 
pcm.justboomdac {
type softvol
slave.pcm "plughw:0"
control.name "Master"
control.card 0
}
pcm.!default {
type plug
slave.pcm "justboomdac"
}
 
##########################
--cat /etc/mpd.conf:
 
music_directory		"/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
playlist_directory		"/home/pi/RPi-Jukebox-RFID/playlists"
db_file			"/var/lib/mpd/tag_cache"
log_file			"/var/log/mpd/mpd.log"
pid_file			"/run/mpd/pid"
state_file			"/var/lib/mpd/state"
sticker_file                   "/var/lib/mpd/sticker.sql"
user				"root"
bind_to_address		"localhost"
auto_update    "yes"
auto_update_depth "10"
input {
        plugin "curl"
}
audio_output {
	type		"alsa"
	name		"JustBoom"
	device		"hw:0,0"	# optional
	mixer_type      "software"      # optional
	mixer_device	"digital"	# optional
}
filesystem_charset		"UTF-8"
id3v1_encoding			"UTF-8"

pi@raspberrypi:~ $ speaker-test

speaker-test 1.1.3

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 128 to 131072
Period size range from 64 to 65536
Using max buffer size 131072
Periods = 4
was set period_size = 32768
was set buffer_size = 131072
 0 - Front Left
Time per period = 0.028163
 0 - Front Left
Time per period = 2.730985
 0 - Front Left
Time per period = 2.729852
 0 - Front Left
[..]

pi@raspberrypi:~ $ mpc ls
ZZZ MP3 Whitespace StartUpSound
ZZZ SubMaster Whitespaces
ZZZ-AudioFormatsTest
ZZZ-MP3-StartUpSound
ZZZ-SubMaster
ZZZ MP3 Whitespace StartUpSound
ZZZ-Podcast-DLF-Kinderhoerspiele
ZZZ-MP3-StartUpSound
ZZZ-AudioFormatsTest
ZZZ SubMaster Whitespaces %RCRSV%
ZZZ-LiveStream-Bayern2
ZZZ This American Life Podcast

martinclausen2, 2019-03-23:
Have you tried in a first step to switch off the GPIO buttons?

iwishitwassummer, 2019-03-23:
Hi,
Thanks for the answer. 
Yes, I did. I’m in contact with the supplier support. Still no resolution. I think I will send it back. 
Thanks again

martinclausen2, 2019-03-23:
mmh strange, just an idea: Try plain installation without phoniebox, try to play test sound?

iwishitwassummer, 2019-03-24:
Yeah, that’s what I tried as well. Not a beep, just a distortion

MiczFlor, 2019-03-25:
Hi @iwishitwassummer 
if this is a hardware issue, could you close the thread?

iwishitwassummer, 2019-04-05:
Hi all,

sorry to be a pain. I ordered the Phat DAC and strangely I've the same issue. I really don't know what to do now. 
It seem as if all is fine, speaker-test, webpage, all seem to be good but I don't get a beep. I've tried multiple headset and speaker. Still nothing. USB audio card works (but its shutting it off now and then, I believe a power problem, anyway).

here's the info. any advice is highly apprechiated.

`##########################
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
##########################
cat /boot/config.txt:
dtoverlay=hifiberry-dac
##########################
--cat /proc/asound/modules:
 0 snd_soc_hifiberry_dac
##########################
--aplay -l:
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
##########################
--aplay -L:
null
    Discard all samples (playback) or generate zero samples (capture)
hifiberry
default
sysdefault:CARD=sndrpihifiberry
    snd_rpi_hifiberry_dac, 
    Default Audio Device
dmix:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dac, 
    Direct sample mixing device
dsnoop:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dac, 
    Direct sample snooping device
hw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dac, 
    Direct hardware device without any conversions
plughw:CARD=sndrpihifiberry,DEV=0
    snd_rpi_hifiberry_dac, 
    Hardware device with all software conversions
##########################
--cat /proc/asound/cards:
 0 [sndrpihifiberry]: HifiberryDac - snd_rpi_hifiberry_dac
                      snd_rpi_hifiberry_dac
##########################
--mpc outputs:
Output 1 (hifiberry) is enabled
##########################
amixer scontrols
Simple mixer control 'Master',0
Simple mixer control 'Software',0
##########################
--cat /etc/asound.conf:
pcm.hifiberry {  
	type softvol  
	slave.pcm "plughw:0"  
	control.name "Master"  
	control.card 0  
}  
pcm.!default {  
	type       plug  
	slave.pcm  "hifiberry"  
}
##########################
--cat /etc/mpd.conf:
music_directory		"/home/pi/RPi-Jukebox-RFID/shared/audiofolders"
playlist_directory		"/home/pi/RPi-Jukebox-RFID/playlists"
db_file			"/var/lib/mpd/tag_cache"
log_file			"/var/log/mpd/mpd.log"
pid_file			"/run/mpd/pid"
state_file			"/var/lib/mpd/state"
sticker_file                   "/var/lib/mpd/sticker.sql"
user				"root"
bind_to_address		"localhost"
auto_update    "yes"
auto_update_depth "10"
input {
        plugin "curl"
}
audio_output {
	type		"alsa"
	name		"hifiberry"
	device		"hw:0,0"	
	mixer_type      "software"      
	mixer_control	"MASTER"		
}
filesystem_charset		"UTF-8"
id3v1_encoding			"UTF-8"
##########################
sudo aplay -v -D plughw /usr/share/sounds/alsa/Front_Center.wav
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono
Plug PCM: Route conversion PCM (sformat=S16_LE)
  Transformation table:
    0 <- 0
    1 <- 0
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 1
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 24000
  stop_threshold   : 24000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1572864000
Slave: Hardware PCM card 0 'snd_rpi_hifiberry_dac' device 0 subdevice 0
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 24000
  stop_threshold   : 24000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1572864000
  appl_ptr     : 0
  hw_ptr       : 0
##########################
sudo aplay /usr/share/sounds/alsa/Front_Center.wav 
Playing WAVE '/usr/share/sounds/alsa/Front_Center.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Mono

martinclausen2, 2019-04-05:
Asking the same question again: Have you tried in a first step to switch off the GPIO buttons?

iwishitwassummer, 2019-04-05:
Yes. I did

martinclausen2, 2019-04-06:
Could you please let us know what you have tried so far?

iwishitwassummer, 2019-04-06:
Ohhh, some many tries... typically I followed first the manufacturer protocol without Phoniebox installation. Then I surfed the internet, later I installed Phoniebox and tried again to play around.
Typically, asound.con and mpd.conf variations like (examples:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/External-Non-USB-Audio-DAC-ES9023 (obviously)
https://forum-raspberrypi.de/forum/thread/42168-phoniebox-dac-installieren-und-anschließen/?postID=372854#post372854
https://nwgat.ninja/pi-zero-as-a-dlna-upnp-render-endpoint-2/ 

Phoniebox's website shows nicely when the volume is not correctly configured.
Let me know your way, I'm ready to start from scratch (which I did several times because of many changes to the conf files).

martinclausen2, 2019-04-11:
@iwishitwassummer mmh, looks fairly complete. My way was first fighting GPIO confusion and then MPD configuration. I documented that in the first link that you cite, so there is nothing I can add at the moment.
In summary you tried many different software installations and two DACs, but always the same RPi. Could it be that you fried some of the GPIOs? Do you have a scope handy to check the signals on the I2S? Do you have a second RPi?

iwishitwassummer, 2019-04-11:
Thanks. That’s currently my conclusion as well. I was considering using my second Pi to test. Problem is, it’s in use. But I might have another chance to test another one. I actually ordered the Pi Zero with soldered heads to avoid this problem. But yeah, things happen. Thanks for helping

martinclausen2, 2019-04-12:
I found the Pi Zero not sufficient for this application. The other box I build with a 3 A+ has a much better usability.

iwishitwassummer, 2019-04-15:
@martinclausen2 thanks. I want to build a portable version. I got hold of a flash drive version of an RFID reader.  I'll try (if I ever solve the DAC problem). Waiting for Pimoroni to comment (or exchange)

closed 2019-04-15 by iwishitwassummer

iwishitwassummer, 2019-04-24:
Just a quick update:
I got hold of another RPi 3 and the DAC worked within 10 minutes. Got the replacement Pi Zero: worked as well. Thanks for helping, I was really belivieving it was me :O