MiczFlor/RPi-Jukebox-RFID #514: Shuffle is not working properly

warlok85, 2019-03-16:
Hi,
first of all, I want to thank you for the great project.  I have installed the Phoniebox on a Raspberry 3B+ using the latest master branch 1.19 with the spotify integration.  

I just want to test the shuffle function but it does not work. Only when I set the second swipe play pause in the web interface and trigger this function with the ID card the shuffle mode work. 
In web interface the shuffle mode is activated all the time.

Any ideas?

saschaschefenacker, 2019-05-22:
+1
great would be to start a random file out of the connected folder each time the card is used on the reader.
But I can't get that to work right now.

After config play/pause the random factor is seen in the playlist

iroQuai, 2019-05-27:
I have the same problem here with the classic version

shinebar1001, 2019-06-24:
I'm using 1.1.9 as well. In my opionion mpc shuffle isn't working properly. I added "mpc random on" in the inc.settingsFolderSpecific.sh - work's for me, even if it's a quite ugly fix:

`    # SHUFFLE FOLDER
    if [ $SHUFFLE == "ON" ]
    then
        mpc shuffle
        mpc random on
    else
        mpc random off
    fi
`

Fl0ete, 2019-09-01:
I'm having the same issue with 1.2-rc1

MiczFlor, 2019-09-29:
Hi, please have a look here: #587 
I think that solves it in the newest master branch. There is a bit of a "thinking around corners" because the logic of shuffle and single play clash a bit. 
Now single play "only" works progressively if you stick to the same playlist. If you come from another list, it will start from the top.

closed 2019-09-29 by MiczFlor

Fl0ete, 2019-10-03:
I'm running 2.0 rc1, but shuffle still doesn't work expected for me.

ZyanKLee, 2019-10-30:
hey, in `develop` this seems not to be fixed at the moment (commit#021570699232552e0eb241a89ef6f410f8313e43).
This is part of the debug.log - with DEBUG=true for `scripts/playout_controls.sh`:

```
########### SCRIPT playout_controls.sh (2019-10-30.08:09:20) ##
  ######### SCRIPT inc.readArgsFromCommandLine.sh (2019-10-30.08:09:20) ##
  # VAR CARDID: 
  # VAR COMMAND: playlistaddplay
  # VAR FOLDER: Giraffenaffen Playlist
  # VAR VALUE: Giraffenaffen Playlist
VAR COMMAND: playlistaddplay
VAR VALUE: Giraffenaffen Playlist
   playlistaddplay VALUE: Giraffenaffen Playlist
   playlistaddplay FOLDER: Giraffenaffen Playlist
  ######### SCRIPT inc.readArgsFromCommandLine.sh (2019-10-30.08:09:20) ##
  # VAR CARDID: 
  # VAR COMMAND: savepos
  # VAR FOLDER: 
  # VAR VALUE: 
#START##### SCRIPT resume_play.sh (2019-10-30.08:09:20) ##
VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
VAR COMMAND: savepos
VAR VALUE: 
VAR FOLDER: 
VAR FOLDER from settings/Latest_Folder_Played: 
  content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders//folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: savepos
  savepos FOLDER: 
#END####### SCRIPT resume_play.sh (2019-10-30.08:09:20) ##
  #START### SCRIPT inc.settingsFolderSpecific.sh (2019-10-30.08:09:20) ##
  # VAR FOLDER from settings/Latest_Folder_Played: 
  # Folder exists: /home/pi/RPi-Jukebox-RFID/shared/audiofolders//folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  # SINGLE TRACK PLAY: OFF
  # # CHANGING: mpc single off
  # SHUFFLE FOLDER: OFF
  # # CHANGING: mpc random off
  #END##### SCRIPT inc.settingsFolderSpecific.sh (2019-10-30.08:09:20) ##
  ######### SCRIPT inc.readArgsFromCommandLine.sh (2019-10-30.08:09:20) ##
  # VAR CARDID: 
  # VAR COMMAND: resume
  # VAR FOLDER: 
  # VAR VALUE: 
#START##### SCRIPT resume_play.sh (2019-10-30.08:09:20) ##
VAR AUDIOFOLDERSPATH: /home/pi/RPi-Jukebox-RFID/shared/audiofolders
VAR COMMAND: resume
VAR VALUE: 
VAR FOLDER: 
VAR FOLDER from settings/Latest_Folder_Played: 
  content of /home/pi/RPi-Jukebox-RFID/shared/audiofolders//folder.conf
CURRENTFILENAME="filename"
ELAPSED="0"
PLAYSTATUS="Stopped"
RESUME="OFF"
SHUFFLE="OFF"
LOOP="OFF"
SINGLE="OFF"
  Now doing what COMMAND wants: resume
  entering: resume with value OFF
#END####### SCRIPT resume_play.sh (2019-10-30.08:09:20) ##
mpc load Giraffenaffen Playlist && /home/pi/RPi-Jukebox-RFID/scripts/resume_play.sh -c=resume
echo  > /home/pi/RPi-Jukebox-RFID/scripts/../settings/Latest_Folder_Played
VAR Latest_Folder_Played: 
```
to me this looks like the value of FOLDER not being available throughout all those includes and script calls.

PR #620 (2019-11-10): fix: handling of FOLDER in playlistaddplay
The code in `inc.settingsFolderSpecific.sh` should read the value for 
`FOLDER` only from `Latest_Folder_Played` if it is not set by its parent 
script.
Otherwise this might return with no value at all for example on the 
first media played and with wrong settings if changing playlists.

fixes #514
commit 57773bf4 (2019-11-10): fix: handling of FOLDER in playlistaddplay

The code in `inc.settingsFolderSpecific.sh` should read the value for 
`FOLDER` only from `Latest_Folder_Played` if it is not set by its parent 
script.
Otherwise this might return with no value at all for example on the 
first media played and with wrong settings if changing playlists.

fixes #514