MiczFlor/RPi-Jukebox-RFID #516: Amplifier power off / standby mode

frickelfilm, 2019-03-19:
Hi there, 
I've set up my RPi-Jukebox on a Raspberry Pi 3 B+ and its going well so far. Thanks to all for your work on this project!

The only issue where I can't get to a solution till now is the power down of the amplifier, when switching off the Raspberry Pi.
I followed the "Hack PAM8403 Amplifier Power Off" from the wiki. But I am using the mono amplifier board from Adafruit: https://www.adafruit.com/product/2130
So I don't have to cut and solder a wire to the standby pin, because there is already one on the board. I've connected it to the GPIO23 and changed this in the "activate-amplifier.py". When I switch the Pin to LOW, the amp turns off - that's good. But when I shutdown the RPi, the amplifier is still on. So I measured the GPIO23 and it still has 3,3V.
The problem is, that in this poweroff/firmware halt mode, my RPi changes automatically to the HIGH (3,3V) state. When I set in "/boot/config.txt" a dtoverlay=gpio-poweroff for the GPIO23, the power off of the amp works well, but I can't start the box again, because the "gpio-poweroff" setting "prevents the ability to boot by driving GPIO3 low", as I read in the documentation...

So has anyone an idea how to fix that? Or am I on a wrong way and have to fix it by adjusting the hardware?

Best, Karl

P.S.: If it is easier to explain, we can switch to german :)

martinclausen2, 2019-03-23:
@frickelfilm Hi, there is a pull-up on the Amp board, it carries the label 1002, remove that one. Instead add a 10k pull-down resistor from GPIO 23 to ground. You may reuse the resistor that you just removed.

frickelfilm, 2019-03-26:
Hi @martinclausen2 
Thanks a lot! Its a big step forward - I've soldered it according to your instructions and now the activation of the Amp board works with the script. 
But: When I boot the RPi, I just hear a short cracking noise from my speaker and then its off again. When I check the status of the "phoniebox-activate-amplifier" service, it says its running.
`sudo service phoniebox-activate-amplifier status`
And when I restart this service, the amp powers on (and turns off when I shutdown the RPi).
`sudo service phoniebox-activate-amplifier restart`
I looked at the config of my RPi, but couldn‘t find a solution till now. As if there was something after the start of the service that turns the GPIO off again.
I am using the standard installation with additional GPIO buttons and the rotary knob.
So maybe anyone has another hint concerning that issue?

Thank you!

martinclausen2, 2019-03-26:
@frickelfilm sounds like a gpio collision - a second script claims the pin and your amp is powered down. Please check, which GPIOs you use for the buttons and the rotary encoder. Please note, that the second encoder for the track has one of its inputs on the GPIO 23. This would exactly explain the effect you observed. Please disable the second encoder or relocate the amp shutdown GPIO. I always recommend the [flexible pin out](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#flexible-pinout) which has no collisions with the encoders, amp, RFID, DAC, etc.

frickelfilm, 2019-03-27:
@martinclausen2 that's it! Thanks a lot! I wasn't aware of the second rotary encoder. I've changed to the flexible pinout and now it works. 
I don't hear the startup sound, because the Amp powers on too late. But thats an issue, which I can fix later. And as I would like to install the status led, there is another indicator, that shows the box is on.

closed 2020-01-18 by MiczFlor