MiczFlor/RPi-Jukebox-RFID #518: After installation of RPi-Jukebox PCM5102a DAC is deaf

nickzam, 2019-03-24:
Hi,
I have a problem with generic chinese PCM5102a module, which looks exactly like this one:
https://www.instructables.com/id/High-Quality-Music-Player-and-Internet-Radio-With-/
I've connected it the same way as it is mentioned in the link, had a clean Raspbian Stretch install, enabled dtoverlay:
dtoverlay=hifiberry-dac in /boot/config.txt and disabled (commented) #dtparam=audio=on .

After that 
`aplay -l` showed correct card:
`**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 []
Subdevices: 1/1
Subdevice #0: subdevice #0`

I can hear output of speaker-test ran like so:
`speaker-test -D default -c 2 -twav`

Then, I download and run install script from github: `stretch-install-spotify.sh`
As audio interface I've put `sndrpihifiberry` and entered Spotify credentials when asked. 
Script succeeds, I can use Jukebox on the web, browse through my Spotify account, but NO SOUND.
`speaker-test -D default -c 2 -twav` plays silence as the Jukebox on the web.
I've read through Troubleshooting section, disabled GPIO-buttons, restarted, had at least 3 clean installs (one with firmware & packages upgrade) from scratch - nothing worked. Sound is always not there after installing RPi-Jukebox and `aplay -l` output is also the same as before installation.

I succeeded once playing some music via DAC & Jukebox when I started installation script second time and accidentally stopped it somewhere in the middle. After that I had no web UI (it returned 404), but I was able to access Iris on port 6080 and was able to play music from Spotify.
Unfortunately, after next clean install this success were gone :(

I've checked through `stretch-install-spotify.sh` script and to my understanding there is not much things that can corrupt working audio interface, at least speaker-test should work fine.
So, I suppose that some of the third-party packages that are being installed somehow corrupt the working state of DAC.

I did an experiment running speaker test in one terminal window and starting `stretch-install-spotify.sh` in second one. Sound from first window were coming all the way through until installation succeeded and I needed to restart Raspberry Pi. 

Now, I run out of ideas how to fix/track which package is destroying the sound, but I'm open to suggestions and ready to help if needed.

hailogugo, 2019-03-30:
Look up in this wiki thread

https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/External-Non-USB-Audio-DAC-ES9023

Suddenly it will works by your phonibox. I have trouble also with my DAC. I think if the startupsound is co0ming, this work by me. Than it will not kill this program and the soundcard is not avaiable for mpd.

nickzam, 2019-03-31:
Unfortunately this didn't worked for me, tried wiki thread on a clean install with working speaker test. 
To experiment, I've started to execute setup steps manually and found out that speaker test doesn't work  after I install mopidy (even before RPi-Jukebox) which is super strange. Any ideas why?
Going to open issue at mopidy's github.

martinclausen2, 2019-04-12:
Do you use GPIO buttons? Please check for collisions in the [wiki](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview#flexible-pinout)

nickzam, 2019-04-20:
Thank you, after I've changed mapping for GPIO buttons everything worked fine. Wiki text is misleading, since it uses phrase "we now use different than classic mapping", but in script uses classic pin mapping :(

closed 2019-04-20 by nickzam