MiczFlor/RPi-Jukebox-RFID #519: Volume level reset to 30 after restart

MontyMcMutton, 2019-03-25:
Hi, I have an issue with volume: I read somewhere that Phoniebox remembers last audio level after restart. 
But no matter what value I setup audio level it‘s always reset to 30 after rebooting.

TM4G1, 2019-03-28:
Hi,

I am facing the same issue. Interestingly the boot sound seems to be at a higher volume.

Best regards
Tom

Prutsium, 2019-03-30:
Same here volume 30% after reboot

Prutsium, 2019-03-31:
I changed in both configs the volume i require and seems to stay at that level.

In: sudo nano /etc/mopidy/mopidy.conf
I changed:
[audio]
output = alsasink
mixer_volume = 70

And in: sudo nano ~/.config/mopidy/mopidy.conf
[audio]
#mixer = software
mixer_volume = 70
output = alsasink
#buffer_time =

Looks like these values are not updated by the software.

TM4G1, 2019-03-31:
Seems to work! Somehow I still have one strange behavior that I thought would be related to the volume. The first card that I am swiping does not play, only if I swipe a second different one it work. After that is runs all fine.

TM4G1, 2019-03-31:
I found that swiping the first card is not the issue. The boot sound just comes to early, if I wait longer with the first swipe after the restart it works fine. So it seems the boot ready sound is not waiting for everything that is required for playing music.

MontyMcMutton, 2019-03-31:
@Prutsium: That means after every reboot volume starts at 70% no matter what value you had set when powering off? 
So the issue basically still remains and you just found a workaround to change the always-the-same start up volume level?
What I did was to manually change the (maxed) volume dial that’s attached to my speakers so 30 percent is not as loud anymore. 

@TM4G1: with my Phoniebox it takes 25 seconds after startup sound until it reacts to the first swipe.

NiMeDia, 2019-04-29:
Having the same issue with the spotify version.

closed 2020-01-18 by MiczFlor