MiczFlor/RPi-Jukebox-RFID #520: Rotary encoder: Reboot on click, wrong order for volume

mawmawmawm, 2019-03-26:
Hi there,
I followed the instructions on how to implement a KY-040 rotary encoder for volume changes.
The instructions worked, wiring was simple, service is running. 
I ran into two issues:
- The knob worked the wrong way: Turning it CW decreased the volume, CCW increased the volume. I was able to change that by switching the [binary commands](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/2a823b496712aeb00a7967384c0b58a91177fe5a/scripts/rotary_encoder_base.py#L31) in `rotary_encoder_base`. So this is no longer an issue anymore, but it's still strange that I had to change this

- When I push the button the RasPi reboots, I'm not exactly sure why. I checked the [wiki](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume) as well as the `rotary-encoder.py` file itself and tried to include a `switchPressed()` but that didn't do anything.

Any help here would be appreciated because rebooting the Pi is really something that I don't want to do when this button is being pushed, ideally it should toggle the volume or don't do anything at all.

Thanks for this project by the way, it's awesome.

mawmawmawm, 2019-04-01:
After a reboot the button click now mutes the volume which is fine. Not sure what changed though.

martinclausen2, 2019-04-12:
fix for the wrong direction had been submitted and merged into dev sometime ago. In order not to break existing installations, the documentation was changed instead of the code. So please revert the changes in the code and swap CLK and Data pins of your encoder to stay compatible with future releases.

Benzinbruder, 2019-12-27:
I ran into the same issue. Funny enough, my test setup worked fine, but now after building the real box, turning the KY-040 CW lowered the volume instead of increasing it.

My first guess was to simply switch the GPIOs assigned to `volU` and `volD` in `gpio-buttons.py`, that did not do anything though.

Thanks to this issue here I had a look at `rotary_encoder_base.py` as well and checked the [encoder state bits](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/2a823b496712aeb00a7967384c0b58a91177fe5a/scripts/rotary_encoder_base.py#L31) as mentioned above.

I found that my code looked like this
```
KeyIncr = 0b00000010
KeyDecr = 0b00000001
```
which is exactly the other way around than in the [current version of the file](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/2a823b496712aeb00a7967384c0b58a91177fe5a/scripts/rotary_encoder_base.py#L31).
I changed the settings to
```
KeyIncr = 0b00000001
KeyDecr = 0b00000010
```
and now setting the volume with the KY-040 works just fine.

By the way, I did a fresh install and followed the [guide for the rotary encoder](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Audio-RotaryKnobVolume) sometime around the beginning of December 2019.

s-martin, 2023-05-09:
GPIO control (since 2.2) completely overhauled the implementation.

Please try to use the new version. If the problem still occurs, feel free to reopen this issue.