MiczFlor/RPi-Jukebox-RFID #522: Enable Google Play Music (GMusic)

bbadaboom, 2019-04-03:
Da ich kein Spotify-Abo habe sondern Google Play Music nutze, wollte ich das entsprechend auch mit der PhonieBox nutzen.

Hier die Anleitung wie man mopidy und damit die PhonieBox um Gmusic erweitert.
Die WebOberfläche habe ich nicht angepasst, aber man kann alles so machen wie mit Spotify (auf der iris-oberfläche bei der Suche oben rechts GMusic aktivieren) und dann bei der Konfiguration der Karten statt die Spotify URI einfach die GMusic URI rein kopieren. Das Cover holt er sich nicht automatisch, aber alles andere scheint zu funktionieren.

**Anleitung:**
Wenn man 2-Faktor-Authentifizierung nutz (sollte man!) muss zuerst in Google für deinen Account ein Application-Passwort eingerichtet werden. Somit musst du auch nicht dein eigenes Passwort in die Konfiguration eingeben und kannst das Passwort und somit den Zugang problemlos jederzeit Sperren. 
Das Passwort wird erstellt indem man auf https://myaccount.google.com/ geht, dann Security --> App Passwords --> (Einloggen) --> “Generate” --> Custon App einen eigenen Namen geben --> Generiertes Passwort merken.

mopidy-gmusic installieren (einer der Schritte hat bei mir ca.10-15min gedauert, also Geduld!):
```
sudo apt-get install libxml2-dev libxslt-dev python-dev
sudo pip install mopidy-gmusic
sudo pip install pyasn1==0.4.1
sudo easy_install pyasn1==0.4.1
```

Passe die Datei `/etc/mopidy/mopidy.conf` an:
`sudo nano /etc/mopidy/mopidy.conf`

Füge die Konfiguration ein:
```
[gmusic]
username = your.name@gmail.com
password = your_password
deviceid = mac
all_access = true
bitrate = 160
```
Falls du nicht "All Access" hast (das Stream-Abo), diesen Punkt in der Config einfach auf false setzen.
#
Mopidy fix für abbrechende Streams ([Issue 161](https://github.com/mopidy/mopidy-gmusic/issues/161))
Einfach `/usr/lib/python2.7/dist-packages/mopidy/audio/actor.py` bearbeiten, so dass:
`_GST_PLAY_FLAGS_AUDIO = 0x02`

So aussieht:
`_GST_PLAY_FLAGS_AUDIO = 0x02 + 0x80`

MiczFlor, 2019-04-03:
Super. Danke. Ich werde das ins wiki nehmen. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

MickaelMa, 2019-04-11:
Work like a charm !

MiczFlor, 2019-05-25:
Hi @bbadaboom 
I made an English version here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Enable-Google-Play-Music-GMusic
Thanks for the input. I will close it for now and call you back on board if somebody encounters problems :)

closed 2019-05-25 by MiczFlor