MiczFlor/RPi-Jukebox-RFID #526: Lieder starten nicht

Eisselpudel, 2019-04-16:
Hallo,
ich habe nach einigen Wochen mein Raspberry wieder rausgeholt und will die Jukebox endlich zum laufen bringen.
Die Installation klappte eigentlich Problemlos. Ich musste die Rechte anschließend einmal ändern, um Ordner erstellen und Dateien hochladen zu können. Gestern habe ich gemerkt, dass ich den MPD Server noch installieren muss. Jetzt kann ich auch die Lautstärke einstellen (obwohl ich nichts höre). RFID Karten werden erkannt und beim VLC-Player kann ich auf der normalen Oberfläche die Lieder abspielen. Nur mit der Jukebox nicht.
Bei Youtube bekam ich den Tipp, die Datenbank nochmal zu scannen. Den Knopf habe ich nicht, oder finde ich nicht.
Anbei zwei Bildschirmfotos von von der Info und dem Reiter "Folders und Files".
Ich meine, hier mal gelesen zu haben, dass man mit einem Befehl noch Systeminfos posten soll. Den finde ich nur nicht wieder.
Wäre super, wenn mir jemand weiterhelfen könnte.
[image]
[image]

MatthiasCzech, 2019-04-22:
Hi, das klingt, als ob du die Phoniebox manuell installiert hast. (btw. Version 1.1.9 ist schon draußen).
Beim installieren mit dem One-Line-Script, wird man nach dem Audioausgabe-Gerät gefragt. Das sollte, wenn man nicht gerade eine USB-Soundkarte oder Hifiberry nutzt "PCM" sein.
Und das wurde dann im Verzeichnis settings in die Datei "Audio_iFace_Name" geschrieben.
prüfe die Datei mal, und falls nicht vorhanden, hoffe ich, es reicht sie einfach zu erstellen und PCM gefolgt von einem Zeilenumbruch reinzuschreiben.
Ich kenne den Changelog jetzt nicht, aber generell würde ich empfehlen mit dem One-Line-Script auf die Version 1.1.9 zu wechseln. (Neu aufsetzen)
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#oneLineInstall

Eisselpudel, 2019-04-22:
Hey Matthias,
du hattest Recht, war eine manuelle Installation. Hatte zwischendurch auch schon auf 1.1.9 geupdatet.
Habe das jetzt mit dem OLS installiert, musste ein bisschen am RFID-Reader rumspielen und es klappt. Voll gut. Danke für die Tipps!!!!

closed 2019-04-22 by Eisselpudel