MiczFlor/RPi-Jukebox-RFID #527: Neuinstallation one liner +Spotify funktioniert nicht

1704eu, 2019-04-18:
Hi all after I tried to setup a new instance of the awesome phoniebox + spotify I haven't been able to get the spotify part working. When I switch to the spotify tab I see an "internal server error"

nickzam, 2019-04-20:
Hi, it worked for me a week ago. Did you entered correct credentials for Spotify? You should copy username from your profile at Spotify, also provide secret key and secret token from mopidy-spotify connection page.
For me install script did not worked from cloned Git repo, so I did installation getting the script via wget and running it.

1704eu, 2019-04-21:
Dear nickzam, many thanks in advance. I guess I found the mistake thanks to your post. I generated a client ID and Secret under https://developer.spotify.com which I then entered. Now I realised, that I should generate a Client ID and Secret from the mopidy website allowing mopidy to access my spotify?
Maybe I am not the only stupid out there ;)
During the installation process you will be asked to enter a client ID and Secret. Go to https://www.mopidy.com/authenticate/ login with your spotify account and take those values from here.
Maybe this would also be something for the installation instructions?

closed 2019-04-21 by 1704eu