MiczFlor/RPi-Jukebox-RFID #529: Feature Request: Content Server

MatthiasCzech, 2019-04-22:
Hi,
I build a box for my niece and my nethew, so I do not live in the same house.
On Easter I had the problem, I prepared 8 cards but they couldn't use them directly.

My  idea is a content server, as stand alone and/or as integration in nextcloud.
At the phoniebox, will be an configuration point whre you can insert link to your server. And an ID for the box.
When the phoniebox don't have a shortcut for the RFID it will login to content server with ID and check wheter there is a zip file for the RFID. Download the file, compare with hash, extract files and connect with RFID.
There should be audiofiles for smal responses, like. "Your playlist will be downloaded, please be patient", "Your playlist is now ready, have fun", "There is no internet connection, inform your parents". "A playlist is currently downloading, please wait"

During download no other download should be possible. It would be good, if it's possible to hear a local playlist during download. Maybe download can be aborted by special RFID command or by pressing and holding a GPIO button, when the new card is the lastUsedRFID

The content server could be created with a small database offering a rest api to check if the box is allowed to get the content and/or as a plugin for nextcloud.

The content server could also be used for a self created upgrade script.
Download a script and execute. (but this should be discussed, because of security risk)

Maybe in July I will find some time to think about a solution, but maybe someone likes the idea and start immediatly.

MiczFlor, 2019-05-04:
Hi @MatthiasCzech 
that is a great idea. I will start thinking about this more around the mid of May, when I have more time again to dedicate to Phoniebox. Right now it's a bit tight with other projects.
Especially the idea of cloning configurations and content is interesting. 
Thanks and all the best, micz

BugHunter2k, 2024-05-23:
I was thinking about something similar. I there any news about this topic? Would be great to have such a functionality for sharing and syncing of audiofiles and configuration.

s-martin, 2024-05-24:
So far nobody has picked up this topic and it's not in any near term planning.

If you are interested in implementing this topic you are more than welcome.

It should be implemented for future3 / V3 and the planned implementation should be discussed here in this issue.

You can also use the chat for quick questions.

AlvinSchiller, 2024-05-24:
Not exactly like described here, but a functionality to sync audio from a server/network share does exist:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Sychnronising-Phonieboxes-in-a-local-network

Maybe this already fits your case.