MiczFlor/RPi-Jukebox-RFID #533: Enable PCSCD-comptable readers

TeGuy, 2019-05-04:
I do have a SCM reader compatible with PCSCD - but it is not able to act as a HID device.

So I implemented a Reader using the RFIDIOt library - and it works!

Have a look at https://github.com/TeGuy/RPi-Jukebox-RFID - it is a 1-hour-dirty-hack, but after cleaning it it may be suitable for core.

TeGuy, 2019-05-04:
One additional comment: I don't know if this makes the RainerSCT Reader working - Back in the days when this reader was a free givewaway I had several problems getting it work with PCSCD. This applies to the touchatag reader, too.

MiczFlor, 2019-05-04:
Hi @TeGuy 
a great example of being your own "[rubber duck](https://en.wikipedia.org/wiki/Rubber_duck_debugging)" developing partner :)
Could you add your contribution to the wiki, please? That would be great!
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki
All the best and share pictures once your Phoniebox is ready,
micz

s-martin, 2023-05-05:
Was implemented in #1250

closed 2023-05-05 by s-martin