MiczFlor/RPi-Jukebox-RFID #534: Spotify Playlists with a lot of tracks, don't work proper

dewenni, 2019-05-05:
Hello
 I tried to use Spotify playlists with a lot of tracks. For example some child stores from Bibi Blocksberg. There are playlists on Spotify that includes all episodes and have often more than 500 tracks. If I try to register such playlist, the Phoniebox get stuck.

To use such playlist would be nice and more easy than register every single episode with separate RFID cards.

so first question:
- Is this a genreal problem, or is this only a problem in my setup?

If it is a general problem, is there any chance to fit it?

MiczFlor, 2019-05-12:
Hi @dewenni 
have you gotten anywhere with this observation? I have no spotify account and always rely on the community to contribute answers.

dewenni, 2019-05-13:
Hi @MiczFlor 
thanks for your reply. No, the situation is still the same, but I haven't done further investigations.
So if there is anybody else with a Spotify account, it would be nice if he can double check this.

closed 2020-01-18 by MiczFlor