MiczFlor/RPi-Jukebox-RFID #536: Security

switzel, 2019-05-06:
One thing that has always bothered me about this project is that the webserver is a sudoer. That seems like the kind of dirty hack that tends to make all these little smart devices so insecure. Although I don't necessarily expect there to be a huge botnet of phonieboxes I wonder: what are the reasons why the webserver needs root permissions?

When I do 

        grep -r sudo *

most uses I see don't really seem necessary. But possibly there is a good reason I'm missing. (Admittedly I don't know how to shutdown without root permissions but I can do it from my GUI so there has to be an established way.)

winterrace, 2019-05-11:
This was contained in the installation manuals up to some point in time:

> The webserver is usually not very powerful when it comes to access to the system it is running on. From a security point of view, this is a very good concept: you don't want a website to potentially change parts of the operating system which should be locked away from any public access.
> 
> We do need to give the webserver more access in order to run a web app that can start and stop processes on the RPi. To make this happen, we need to add the webserver to the list of users/groups allowed to run commands as superuser.

MiczFlor, 2019-05-12:
Hi @switzel @winterrace 
the root for the webserver is still from "day one" and I am sure can be written out of the code - eventually. I tried it once and then it broke and I didn't have the time to go through this one by one. And then reverted.
At the end of the day I stick to what @winterrace quotes. In essence: don't use Phoniebox code on a live server ever. Keep it in the family :)

closed 2019-05-12 by MiczFlor

switzel, 2019-05-15:
Hi @MiczFlor 
Sure but as it is it takes a bug/misconfiguration of lighttpd and a bug/misconfiguration of the router to completely compromise the average phoniebox (as opposed to allowing to serve funny content). It's like a shortcut through all security layers from the most exposed to the core. I do understand that you don't like to invest time to work on this and I obviously I don't mean to tell anyone what to do. But I don't exactly consider this issue as "closed". Perhaps one day I'll give it a try.