MiczFlor/RPi-Jukebox-RFID #539: Problem mit Störgeräuschen

TeGuy, 2019-05-09:
Hallo zusammen,

ich habe an meinem RPi aktive USB-Lautsprecher von Logilink. (Für sagenhafte 5€). Blöderweise bekomme ich nun Störgeräusche, wenn ich z.B. auf die Weboberfläche zugreife (aber auch im idle). 

Die Störgeräusche gibt es sowohl über den internen Klinkenausgang, als auch über die USB-Soundkarte.

Habt ihr eure Aktiv-Boxen irgendwie abgeschirmt oder besonders geerdet? Gibt's einen einfachen Trick?

Beste Grüße
Philipp

domu83, 2019-05-09:
Hi!
Also ich hatte Anfangs mit dem internen Klinkenausgang auch Störgeräusche und nutze den Entstörfilter von Aukey: https://www.amazon.de/dp/B01LX0H29W/ref=cm_sw_em_r_mt_dp_U_8fl1CbZEWJ16H 
Mittlerweile auch an der USB Soundkarte im Einsatz und ich bin zufrieden.

winterrace, 2019-05-11:
Wie Du schreibst, bekommen Deine USB-Lautsprecher den Sound über Klinke, nutzen also den USB-Anschluss offenbar nur für die Stromversorgung (wie z.B. auch die Trust Leto).

In diesem Fall sollte es helfen, wenn Du die Boxen in eine andere USB-Buchse steckst, Hauptsache nicht eine vom Raspi. Dann sollten die Störgeräusche Geschichte sein - jedenfalls hat der Tipp schon einigen anderen Usern auf Anhieb geholfen.

Ich habe in meiner Box zusätzlich ein altes Netzteil von einem alten Smartphone verbaut - soweit keine Störgeräusche :)

MiczFlor, 2019-05-12:
@TeGuy 
Auf der Startseite habe ich einen Link zu dem Ground Loop Isolator / Entstörfilter Audio 
https://github.com/MiczFlor/RPi-Jukebox-RFID#special-hardware
Bei mir funktioniert der bestens.

closed 2019-05-12 by MiczFlor