MiczFlor/RPi-Jukebox-RFID #541: Feedback: Folders are automatically linked to cards when folder name matches card ID

topas-rec, 2019-05-18:
I made a fresh install (incluing Raspbian) and just put my folders on the Phoniebox - no card linking. My folder names are named according to their belging card IDs.
As I swiped the first card to link it, the correct folder already played. I thought well I just changed to another universe. To test that the universe is the same I swiped a card that was linked to one of my initial folders that are not named according to card ID. That one didn't played. OK, linked it using the web app... worked. Universe is still the same... :-)

That might be useful (if not already discovered) for complete fresh reinstalls (no relinking needed) and initial preparation of the box (linking can be done in advance on a PC by reading the ID from the card and name the folders)

MiczFlor, 2019-05-25:
Hi @topas-rec 
the folder linking to the card ID, that's correct. It's not in the wiki, true. When registering a new card, the content of the shortlink is set to the card ID. Then, later, it can be edited.

topas-rec, 2019-05-27:
I would strongly suggest to not only add this to the wiki, but to mention this in the manual or installation as best practice. Hiatus a time saver when your Phoniebox is dead by software and a reinstall is the only option.

Reinstall, copy folders to specified location and everything is up and running again.

topas-rec, 2019-05-27:
ID cards even have the number printed on them mostly. So it is a good connection between folder and card (and its cover/album art of course)

dnmtch, 2019-08-29:
A little hint I'd like to have on the Phonibox . (not that I have the feeling to be allowed to wish here!)

**Ideally Subfolders should be started directly via the Card-ID number:** (card swipe)

Example:

"Folder Name"
- "01234567890" (Subfolder 1)
- "11234567890" (Subfolder 2)
- "21234567890" (Subfolder 3)
"Folder Name Folder 2"
-  "91234567890" (Subfolder 1)
-  "81234567890" (Subfolder 1)

So larger collections of music are easily possible on the Phonibox without overloading the web interface. Nevertheless the content can be recognized by the folder name. The function of the folders (incl. subfolders) can remain the same.

topas-rec, 2020-06-02:
I've mentioned this as a note under https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#making-a-human-readable-shortcut-for-a-card

closed 2020-06-02 by topas-rec