MiczFlor/RPi-Jukebox-RFID #542: Power off does not work

tmesza, 2019-05-26:
Hello together,
i want to ask if somebody can help me with my power off issue?
I build the Jukebox with the instructions from MiczFlor and a Pi3B+, thanks for sharing that great work. The kids are like the box :)
Sometimes is it not possible to power off the box by using the on/off button, the red led on the RPI is showing a red light. I use an on/off shim. I can only power off the RPI by remove the Power by the USB-Power connection.
thanks in advance for your help.
tmesza

domu83, 2019-05-28:
same problem on my setup as well, no resolution found so far

RogoTe, 2019-06-04:
Might be a problem with your onoff-shim. The problem is that without such shim you cannot power down a RaspPi completly. The red LED stayes on although the SoC is down.
So my best guess is, that the shim somehow did not do it's job.

s-martin, 2020-03-07:
@splitti provided an extensive explanation regarding onoffshim in [his blog](http://splittscheid.de/selfmade-phoniebox/#3A), but it’s in German.

closed 2020-12-18 by s-martin