MiczFlor/RPi-Jukebox-RFID #543: Tip for headless installs (without booting to Desktop systems)

Prutsium, 2019-05-29:
Today i thought let's re-install our Phoniebox as i had some issues that Mopidy started quite slow (long loading time)
Since it was a fresh install i decided to install the "Lite" version of Stretch as i dont need the desktop anyway.
All worked fine but i noticed one thing at the end: low volume level of the aMixer

Would be good to include in the install script the following line: sudo amixer cset numid=1 100%
As then the volume will be set to max and volume control of Phoniebox is at default 30%
(When this line is not set the 30% of Phoniebox relates to about 10% level)

MiczFlor, 2019-06-07:
Hi @Prutsium 
I will add this to my `develop` branch and see if it works. Thanks for the heads up!
micz

MiczFlor, 2019-06-07:
I added it as you can see here: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/1fb92db2228da87c2dc1fd95b3514ffec5789466
Thanks for the hint. I will close this once develop is master

Prutsium, 2019-06-07:
Might be related to this but The Audiomixer and PCM volume control are not linear.
Means 60% on audio mixer is 0% on the PCM.

This results in very small headroom for volume control (only 20% is the range from 0 - 100)

Any experience on this?

Prutsium, 2019-06-07:
@MiczFlor see here the issue:
PCM is at 0db but Mixer at 50% so bandwidth for control is very limited.
(MPD Version with Analog audio from jack on Pi)
[image]

Prutsium, 2019-08-24:
Still struggling with this issue as i have very limited area to control the volume.
60= level 0 and 92= Max what makes volume control difficult.
Mixer = PCM so no clue why this happens

closed 2019-09-13 by MiczFlor