MiczFlor/RPi-Jukebox-RFID #544: Hifiberry DAC+ Sound not working

highwaystar42, 2019-05-31:
I don't get the sound of my soundcard Hifiberry DAC+ working.
Raspbian GNU/Linux 9.3 (stretch)
Plus Edition (feat. Spotify Integration) 1.1.9

At the first installation without spotify it works fine.
Just had to set "Master" instead "PCM" in /etc/mpd.conf

After that I did a fresh installation with spotify. Now it does't work.
Only when I reboot the Raspi, I can hear the starting-sound.
As I read, mpd is deaktivated in Spotify-Edition I didn't edit anything in /etc/mpd.conf.

The Spotify Installer only asks for the iFace Name. I tried various entries (Master, PCM, Digital, Analogue) but it's not working.
How do I get the right iFaceName?
Where do I have to set up the soundcard? Only in "settings/Audio_iFace_Name"?
How can I test the sound?

amixer scontrols shows:
`Simple mixer control 'DSP Program',0
Simple mixer control 'Analogue',0
Simple mixer control 'Analogue Playback Boost',0
Simple mixer control 'Auto Mute',0
Simple mixer control 'Auto Mute Mono',0
Simple mixer control 'Auto Mute Time Left',0
Simple mixer control 'Auto Mute Time Right',0
Simple mixer control 'Clock Missing Period',0
Simple mixer control 'Deemphasis',0
Simple mixer control 'Digital',0
Simple mixer control 'Max Overclock DAC',0
Simple mixer control 'Max Overclock DSP',0
Simple mixer control 'Max Overclock PLL',0
Simple mixer control 'Volume Ramp Down Emergency Rate',0
Simple mixer control 'Volume Ramp Down Emergency Step',0
Simple mixer control 'Volume Ramp Down Rate',0
Simple mixer control 'Volume Ramp Down Step',0
Simple mixer control 'Volume Ramp Up Rate',0
Simple mixer control 'Volume Ramp Up Step',0`

MiczFlor, 2019-06-07:
Hi @highwaystar42 

there is a section in the troubleshooting that might help?
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#hifiberry-dac-soundcard-details

And a more general section about "Audio not working":
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working

I hope this helps. If you figure it out, please post your solution here before you close the issue, so others can benefit from your experience.
Thanks and all the best, micz

highwaystar42, 2019-06-07:
Thanks for your advice.
I'll try again after holidays.
After a bit more research I think it's a problem with GPIO.
If i have the solution i will tell.

lj2000, 2019-06-09:
Hi @highwaystar42,

I agree with you. The problem is the original GPIO pinning.
I use the hifiberry miniamp, which conflicts with the originial pinning of the buttons.
So I had to change the pinning at the hardware and also in the gpio python script

Regards
lj2000

highwaystar42, 2019-07-01:
Okay. The problem was the GPIO Pins.
In /scripts/gpio-buttons.py i commented the pins out the Hifiberry i using. See: [https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview]

The soundcheck `speaker-test -Dhw:0,0 -c2 -twav` is working now.

But in the Webfrontend sound is not working anyway. There might be another problem.
I think the problem is somewhere between mpd and mopidy.

I haven't understand yet in which way mpd was replaced by mopidy in the spotify version.

saschaschefenacker, 2019-07-07:
I have the same issue.
I started with 2019-06-20-raspbian-buster-full
but here the installation did not work at all.

Shifted to 2019-04-08-raspbian-stretch
After installation of raspbian I activate the DAC
curl https://get.pimoroni.com/phatdac | bash
reboot und run same command again does start the auto - test -- working fine

Then installing the Jukebox
cd; rm stretch-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/stretch-install-default.sh; chmod +x stretch-install-default.sh; ./stretch-install-default.sh

As audio I used "Master" rest is pretty default.

After that the webForntend works - but no volume and of course no "play"
But also the audiotest stopped to work :(

I'll try some progress as soon as possible
But right now I'm lost.

Also tried
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#hifiberry-dac-soundcard-details

Hardware is:
https://www.aliexpress.com/item/Raspberry-Pi-pHAT-Sound-Card-I2S-interface-PCM5102-DAC-Module-24-bit-Audio-Board-With-Stereo/32742608325.html

 cat /proc/asound/modules
 0 snd_soc_hifiberry_dac

Next step would be the downgrade to the old image - even I do not expect a change with that.

s-martin, 2020-04-12:
Does this issue still persist?

s-martin, 2020-05-31:
If the issue still persists, please reopen the issue.

closed 2020-05-31 by s-martin