MiczFlor/RPi-Jukebox-RFID #545: Youtube DL not working anymore

Yordan1976, 2019-06-01:
It used to work fine before, but not anymore and I don't know how to investigate the reason.

I created a folder \Einzeln\Lied_über_Tiere and used the GUI to connect a card to this youtube file: https://www.youtube.com/watch?v=0ARm654fM_A

The page already reloads after a few seconds but nothing was downloaded. When it worked, the page realoaded after download and conversion.

youtube-dl.log is written with this content:
[youtube] 0ARm654fM_A: Downloading webpage
[youtube] 0ARm654fM_A: Downloading video info webpage

I also tried it with other folders (no subfolder, no Umlaute, created by hand, created by the GUI).

Any hints? Does it still work in general?
What would be the console command to do this? Maybe I would see error messages then. Or is there another logfile to have a look at?

MiczFlor, 2019-06-07:
Hi @Yordan1976 
Hopefully this will answer your questions:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#fix-broken-youtube-dl-while-staying-on-same-rpi-jukebox-rfid-releaseversion
All the best, micz

Yordan1976, 2019-06-07:
Perfect answer, thanks a lot!
I read so much in the wiki but never saw this :)
After the YT update, everything works fine again!

closed 2019-06-07 by Yordan1976