MiczFlor/RPi-Jukebox-RFID #546: what do the "resume", "shuffle" and "single" buttons do?

iroQuai, 2019-06-03:
Okay i think i can guess most of it, but I can't find it in the wiki....

This is what i figured out:
- Resume: the content of this folder starts from the time where it was last stopped. (great for audiobooks or podcasts, i guess)
- Shuffle: the tracks in this folder are played in a random order (so you'll need at least two tracks in the folder?)
- Single: I don't know this one. Is the content of the folder only played one time, instead of repeat-all that are in this folder? 

Thanks in advance for this maybe easy and simple question...

Yordan1976, 2019-06-04:
Single stops after every track. For example for a guessing game with animal sounds. Also a good choice for audio books.

MiczFlor, 2019-06-21:
Hi @iroQuai @Yordan1976 
thanks for clarifying this. I added your writing to the Manual here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/MANUAL#manage-playout-behaviour
(fun fact, I thought I had this all covered. But it looks as if it is time to update the manual. Many things were added since I last touched it...)

closed 2019-06-21 by MiczFlor