MiczFlor/RPi-Jukebox-RFID #547: cant change volume in webapp

iroQuai, 2019-06-03:
I can't change the volume in the webapp. It's always grayed out at 0%. Audio does work, by the way. 

setup:
- used a clean install raspbian stretch lite
- installed phoniebox V1.1.9 classic via the one-line code. 
- i use a Pi3 with [this USB-DAC](https://www.adafruit.com/product/1475). 

Tried all the tips in the troubleshooting FAQ. I double checked my configuration: 
- used `cat /proc/asound/modules` to check that  my USB DAC is 'card 1'
- changed defaults.ctl.card and defaults.pcm.card both to 1 in `/usr/share/alsa/alsa.conf`
- changed `/etc/asound.conf/` to card 1 (like in the wiki/FAQ)
- used `amixer scontrols` to check the right iFace name (it's 0)
- changed the iFace name inside `settings/Audio_iFace_Name` and `/etc/mpd.conf`

rebooted, still the same problem. Interesting thing is that i can edit the maximum output volume with `alsamixer -c 1` ([as explained here](https://learn.adafruit.com/usb-audio-cards-with-a-raspberry-pi/setting-audio-levels))

Does anyone have an idea how I can enable the volume change in the web app?

MiczFlor, 2019-06-07:
Which version are you running and on which branch? The `develop` branch of the latest version has a couple of tweaks I haven't tested yet.

iroQuai, 2019-06-07:
I just used the one line install command as found in the wiki:

cd; rm stretch-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/stretch-install-default.sh; chmod +x stretch-install-default.sh; ./stretch-install-default.sh 

I guess I'm on the master branch...

MiczFlor, 2019-06-21:
Have you had a look at this trouble shooting guide?
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working
I assume you are right to guess that you need to find the correc `iFace` name inside `settings/Audio_iFace_Name` and `/etc/mpd.conf`.

"How do I get the right iFace name?", you might wonder. 

type `amixer scontrols`

iroQuai, 2019-06-21:
I've tried everything in the wiki/faq, as I've tried to mention in the  starting post. Is there something I've maybe missed?

iroQuai, 2019-08-20:
I'm sorry I've triple-checked everything, but I really think I've done everything like the wiki tells me too. I still can't change volume since it's always at "-1%". Although I can edit the max volume, changing this setting doesn't have effect on the output volume. It looks like this in the settings. 

I really hope someone can help me! 

[image]

kerki, 2019-08-22:
What is your output of `amixer scontrols`?

iroQuai, 2019-08-22:
The full output is as follows:
Simple mixer control 'Speaker',0
Simple mixer control 'Mic',0
Simple mixer control 'Auto Gain Control',0

kerki, 2019-08-22:
You have to use `Speaker` in both config files `settings/Audio_iFace_Name` and `/etc/mpd.conf` as your iFace name.

iroQuai, 2019-08-22:
YESS it works now! 
Thanks a lot. Silly me, I thought the part between the '  was the name of the subject and the part behind the comma was the real name. Anyway, thanks a lot for the help!!

closed 2019-08-22 by iroQuai

MiczFlor, 2019-08-23:
Hi @iroQuai @kerki 
this was actually a difficult one to catch :) I used your conversation to make the wiki better. I added your example here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#install-mpd-the-music-player-daemon

where it now says:

**Maybe you need to change the audio iFace ("yourAudioIfaceNameHere") in the config file, too.** By default it uses `PCM` which should work out of the box. If it doesn't, type in terminal:

~~~
amixer scontrols
~~~

This should create an output similar to this:

~~~
Simple mixer control 'Speaker',0
Simple mixer control 'Mic',0
Simple mixer control 'Auto Gain Control',0
~~~

In which case, instead of `PCM` you need to use `Speaker`.