MiczFlor/RPi-Jukebox-RFID #548: Revert from Spotify edition back to MPD

Prutsium, 2019-06-04:
Hi all,

I was running for a while the Spotify version but as i have some issues there i wanted to revert back to the MPD version (Classic) but somehow it keeps playing over Mopidy.

I just run the Classic version of the script and thought all would be set back to MPD in that case but seems not (or i did something wrong :) )

The reason i want to revert back to MPD version is simple: 
- When system just started it takes very very long before first card is playing (Internal connection to Mopidy seems to be very slow)
- Mopidy is not willing to work with the resume function (or has this to do with the naming issues?)
- When system is not used for 10 minutes or so playing a card again takes very long (seems again to be internal Mopidy connection issue)

I am hoping all of this would be solved by reverting back to MPD but system is stuborn and keeps using Mopidy.

Any tips?

MiczFlor, 2019-06-07:
Hi @Prutsium 
once you figured it out, please add it to the wiki :) I am unsure myself. Simply running the install script for the non-spotify version (Classic) does not work, that's true. On startup modipy is initialised. Manually uninstalling mopidy might do the trick and then running the script again.
The install script should detect the existing install and allow you to keep audio files and configurations. Did that work?
All the best, micz

Prutsium, 2019-06-07:
I ended up on a fresh install :)
Loading time is now really WHOW! 15 seconds from power on the card is read and playing.
With the Mopidy version this could be up to 2 minutes (!)

JuliH29, 2019-12-23:
> I ended up on a fresh install :)
Just to be clear: @Prutsium did you end up with a clean install of Buster or Phoniebox?