MiczFlor/RPi-Jukebox-RFID #549: WM8960 Hi-Fi Sound Card HAT for Raspberry Pi

dnmtch, 2019-06-07:
Hi

I tried my luck with a WM8960 Hi-Fi Sound Card. Unfortunately the sound doesn't work anymore after installing RPi-Jukebox.

- User Manual: https://www.waveshare.com/wiki/WM8960_Audio_HAT
- Install Driver / Test Sound (working)
- Install RPi-Jukebox-RFID (using 'Speaker' as Audio_iFace_Name ???)
- Sound not working (startup sound works, so MPD settings must be wrong?)

[image]

More: http://www.raspberrypirobotics.com/setting-up-the-seeed-studio-respeaker-2-mics-pi-hat/

MiczFlor, 2019-06-07:
Hi @dnmtch 
once you figure out how to make it work, please post the result here for others to learn from.
You say "startup sound works, so it might be MDP settings". Have a look in the wiki here: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working 
The section "b) Audio worked, then you changed the sound card and it stopped working" might give you some good pointers. 
All the best ,micz

lj2000, 2019-06-09:
Hi @dnmtch 

I had the same symptoms/problem when I set up my phoniebox which uses the Hifiberry miniamp.

Quick test: 
---
You can try to disable the gpio systemd-service (buttons) and restart the system.
After that your soundcard should play all the time.

Deeper analysis:
---
Normally the systemd-service for the startup sound starts before the gpio-service. So the startup sound can be played without any problem.
The GPIO service configures the hardware-pinning of the button's-GPIOs and listens to these pins after that.
In case of the Hifiberry miniamp and possibly in your case there is a hardware conflict between the original button's pinning and the sound card.

If there is a hardware conflict in your setup I can provide 2 solutions to solve the problem:

- If you want to use buttons in your setup modify the original gpio python script
- If you don't need the buttons disable the service for the gpio buttons

Regards 
Lj2000

belinea4071, 2019-06-16:
I struggled with the same problem on this hardware.
When the raspi is booting up, it does work like a charm, but not when the Website is online. There is no Sound on the speaker.

Thanks to @lj2000, this did the trick for me. 

My solutions so far, I changed the following lines on “gpio-buttons.py”

sudo nano /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py

#shut = Button(3, hold_time=2)
#vol0 = Button(13,pull_up=True)
#volU = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
#volD = Button(19,pull_up=True,hold_time=0.3,hold_repeat=True)
#next = Button(26,pull_up=True)
#prev = Button(20,pull_up=True)
#halt = Button(21,pull_up=True)
#reco = Button(6, pull_up=True) # Choose GPIO to fit your hardware
#play = Button(12,pull_up=True) # Choose GPIO to fit your hardware

On the installation I used AUDIOiFace="playback".

Now it is playing and hear the music as well.

sudo nano /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py

#shut = Button(3, hold_time=2)
#vol0 = Button(13,pull_up=True)
#volU = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
#volD = Button(19,pull_up=True,hold_time=0.3,hold_repeat=True)
#next = Button(26,pull_up=True)
#prev = Button(20,pull_up=True)
#halt = Button(21,pull_up=True)
#reco = Button(6, pull_up=True) # Choose GPIO to fit your hardware
#play = Button(12,pull_up=True) # Choose GPIO to fit your hardware

On the installation I used AUDIOiFace="playback".

Now it is playing and hear the music as well.

dnmtch, 2019-06-18:
**Thank you so much!** Your solution worked. Now I will try to make the button on the soundcard work.

MiczFlor, 2019-06-21:
Hi @dnmtch @lj2000 @belinea4071 
thanks for your input, I added the content of this thread to the wiki here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Sound_Card_WM8960_Hi-Fi_HAT
Could you please check if this is correct? (it's mainly copy and paste :)

One question @belinea4071 
when you say you "changed the following lines on “gpio-buttons.py”
it's not clear from what to what. Now it looks as if you commented all the lines out?

closed 2019-06-21 by MiczFlor

dnmtch, 2019-07-10:
`it's not clear from what to what. Now it looks as if you commented all the lines out?`

**yes**, all the lines are commented out. (for me) the soundcard is only working if  the service for the gpio buttons is disabled and/or the script (gpio-buttons.py) is modified.

i think/hope that with the right Hardware Pinout Assignment there should be a solution:
[image]

I2S_LRCLK on GPIO 19 / Pin 35 vs. MCLK etc.

I need to learn how this Pinout Assignment works. Maybe someone has the knowledge and the time to give a hint or solution.

| GPIO | Pin | WM8960 Hi-Fi Sound Card |
|------|-----|-------------------------|
| 2    | 3   | I2C_SDA P_SDA  |
| 3    | 5   | I2C_SCL P_SCL  |
| 8    | 24  |   |
| 9    | 23  |   |
| 10   | 19  |   |
| 11   | 21  |   |
| 16   | 36  |   |
| 18   | 12  | I2S_CLK P18  |
| 19   | 35  | I2S_LRCLK P19  |
| 20   | 38  | I2S_ADC  |
| 21   | 40  | I2S_DAC  |
| 24   | 18  |   |
| 25   | 22  |   |
| 26   | 37  |   |

michnixweiss, 2020-04-22:
Hey, this is my first post here (but my second phoniebox) and I'm not sure if it's right here!? Please correct mehr (and my english) if necessery.
Now my question/problem:
I'm using the waveshare wm8960 HAT on a raspi zero. Got it working with the wiki except one thing, I cannot figure out how to use the headphone jack detection. Means how can I mute the speaker if a headphone is plugged.
Many thanks

dnmtch, 2020-04-22:
Hi @michnixweiss 

To set sound and adjust volume, you can use alsamixer tool
`sudo alsamixer`

Try to "hear" sound on the speakers, on headphone jack and "both"? I am not sure, if the HAT supports auto headphone jack detection (it is not indicated in the manual).

do you use GPIO buttons/rottary?

michnixweiss, 2020-04-22:
@dnmtch
I'm using gpio and rotary, all working (hat to switch some pins). There is sound on speaker and headphone and every single is adjustable with alsa. But the "normal" cut off/mute the speaker when headphone is plugged is not working. So the headphone (to listen at your own) is useless.
In the datasheet on page 45 the jack detection is mentioned, but I don't have enough knowledge to unterstand how that has to be done. Is it a hardware switch or a software switch? If hardware I think mine is broken, if software does that have to be implemented in the driver, in alsa or done by a script which listens to a gpio?

michnixweiss, 2020-04-22:
The wiki is a little bit problematic from my point of view. Because there is different hardware used in multiple combinations so there is no "general" pinout possible. I think everybody has to check which pins are in fix use (like this HAT "blocks" GPIO 2,3,18,19,20,21) and which can be changed. In my solution rotary uses (out of the box) GPIO 5,6 so I pinned it just on to of the HAT and the rest has to be switched in the gpio-buttons.py
Be careful with not mixing up GPIO and PIN numbers!

But if I understand you correct, you have the same headphone jack behaviour!? Just want to know if the hardware isn't damaged.

dnmtch, 2020-04-22:
> you have the same headphone jack behaviour!?

I have sound on speaker **AND** headphone jack (connected).

i tried to play with alsamixer, searching online etc. i think/hope this should be solvable with asound.conf oder mpd.conf. But here ends already my limited knowledge. sry ;-)

if you find time, just "paste" your build (wiring) in the wiki as a "working solution" others (like me) are going to just copy your approach.

michnixweiss, 2020-06-25:
Unfortunately I got no feedback to my [issue on waveshare github](https://github.com/waveshare/WM8960-Audio-HAT/issues/9).
BUT: I found a solution. Granted a "ugly" one. I found out that the first right pin of the headphone jack is pulled to ground if one is plugged. So I wired this contact to pin 37 (GPIO 26) and wrote a little script which is watching this GPIO, turns off the speaker if it is pulled to ground.
I can make a sort of tutorial if someone is interested;-)
[image]

mistersunshine2020, 2020-12-19:
Hi, I'm building a new phoniebox setup using this soundcard. I've followed all the relevant instructions in the wiki, but I'm not getting any volume control on the web interface. Should I be able to control the volume using the web script at this stage? I'm just about to install a rotary encoder which I'm concerned won't make any difference if there isn't software enabled volume control.

All elements work - GPIO buttons, RFID reader, external hard drive with albums, and 16x2 LCD screen. This is the last step before assembling it as a Christmas present. 

The one thing I couldn't do is this step in the wiki, which I suspect might be the issue: 

**you **must ** disable the service for the gpio buttons or modify the original gpio python script sudo nano /home/pi/RPi-Jukebox-RFID/scripts/gpio-buttons.py**

I couldn't find that file in that location. 

Any help much appreciated! 

Best,

Tom

mistersunshine2020, 2020-12-19:
> Unfortunately I got no feedback to my [issue on waveshare github](https://github.com/waveshare/WM8960-Audio-HAT/issues/9).
> BUT: I found a solution. Granted a "ugly" one. I found out that the first [...]

I would love a bit more information about headphone's working and disabling speakers once I can get the rotary encoder and volume control issue sorted!

s-martin, 2020-12-19:
GPIO is done since 2.1 via [GPIO control ](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control)

configuration is done via `./settings/gpio_control.ini`

mistersunshine2020, 2020-12-19:
> GPIO is done since 2.1 via [GPIO control ](https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/develop/components/gpio_control)
> 
> configuration is done via `./settings/gpio_control.ini`
> 
> I would recommend to open a new issue, if you have problems. Comments in closed issues are easily overlooked.

Thanks. I had edited that file for the GPIO configuration, but will open a new issue for the software controls for volume. 

Much appreciated.