MiczFlor/RPi-Jukebox-RFID #551: KKMoon HID 413d:2107

heikon84, 2019-06-10:
Hallo zusammen,
ich habe dieses Projekt gefunden und finde die Idee echt super. Das ist mein erstes Projekt in dieser Art.
Issue #368 habe ich gelesen und leider brachte mir leider keinen Erfolg.
Issue #327 von cgnmause hat mir leider auch nicht geholfen.

Ich habe mit den Reader mit der HID 413d:2107 geholt und ich drehe mit dem Ding bald durch.
Die Installation ist heute erfolgt und ich habe den aktuellen Master genutzt. 
Ich habe die Reader.py gegen die Reader.py.kkmoonRFIDreader ausgetauscht. 
Im Terminal oder Texteditor funktioniert der RFID Reader, aber in der Phoniebox kommt leider nichts an und es wird keine lastedID.txt geschrieben. 
Weiterhin crashed der Phoniebox RFID-Reader Service.
● phoniebox-rfid-reader.service - Phoniebox RFID-Reader Service
   Loaded: loaded (/etc/systemd/system/phoniebox-rfid-reader.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2019-06-10 18:39:47 CEST; 42min ago
  Process: 10325 ExecStart=/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py (code=exited, status=1/FAILURE)
 Main PID: 10325 (code=exited, status=1/FAILURE)

Jun 10 18:39:46 jukebox systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.
Jun 10 18:39:47 jukebox systemd[1]: phoniebox-rfid-reader.service: Service hold-off time over, scheduling restart.
Jun 10 18:39:47 jukebox systemd[1]: Stopped Phoniebox RFID-Reader Service.
Jun 10 18:39:47 jukebox systemd[1]: phoniebox-rfid-reader.service: Start request repeated too quickly.
Jun 10 18:39:47 jukebox systemd[1]: Failed to start Phoniebox RFID-Reader Service.
Jun 10 18:39:47 jukebox systemd[1]: phoniebox-rfid-reader.service: Unit entered failed state.
Jun 10 18:39:47 jukebox systemd[1]: phoniebox-rfid-reader.service: Failed with result 'exit-code'.

Rasbian ist auch aktuell.
lsusb=> Bus 001 Device 013: ID 413d:2107
Über "ls -la /dev/input/by-id/" wird der Reader ebenfalls angezeigt.

Kann mir da jemand helfen?

korbinianfritsch, 2019-08-01:
Hallo Heiko.
Ich habe gerade das selbe Problem gehabt: der Service läuft bis man eine Karte scannt und failt dann wie von dir beschrieben.

Ich konnte das Problem beheben, indem ich in `/scripts/daemon_rfid_reader.py`, Zeile 16 wie folgt ersetzt habe:
```
        cardid = reader.readCard()
```
statt
```
        cardid = reader.reader.readCard()
```

Der Fehler war, wie ich dem Output von `journalctl` entnehmen konnte, dass im File `/scripts/daemon_rfid_reader.py` die `cardid` mittels `reader.reader.readCard()` versucht wurde zu lesen. Die Reader-Instanz von `Reader.py.kkmoonRFIDreader` hat aber (im Gegensatz zum default `Reader.py`) kein Attribut `reader`.

Ich hoffe, das hilft!

MiczFlor, 2019-08-02:
Hallo @korbinianfritsch @heikon84 
Ich würde den Hinweis sehr gerne im Skript mit reinschreiben. Selber habe ich kein KMOON Reader, kann es also nicht testen.
Ich habe das im `develop` branch als kommentar mit reingeschrieben. Stimmt das so? Bitte mal nachschauen und @heikon84 : bestätige doch bitte hier, ob dein Problem mit dem Hinweis gelöst ist. 

Das geänderte File ist hier: 
https://github.com/MiczFlor/RPi-Jukebox-RFID/commit/bbbf23a815616e39ef6447dc4d97106f66e1c4b1

korbinianfritsch, 2019-08-02:
Hi @MiczFlor.
Danke für die Initiative. Ich würde ggf. noch einen Hinweis in die `Reader.py.kkmoonRFIDreader` schreiben, dass man das beachten sollte, wenn man damit die default `Reader.py` ersetzt.

heikon84, 2019-08-11:
Hallo,
danke @korbinianfritsch der hat wunderbar funktioniert.
@MiczFlor kann man an den Einrichtungsprozess eine if-Abfrage einbauen, sodass die Reader.py und die daemon_rfid_reader.py entsprechend ausgetauscht bzw. angepasst wird. So könnte man wahrscheinlich die manuell Anpassung nicht mehr nötig machen. 
Mein gedachten Ablauf:
wenn HID = 413d:2107 dann
          cp Reader.py Reader.py.default
          rm Reader.py
          cp Reader.py.kkmoonRFIDreader Reader.py
          cp daemon_rfid_reader.py daemon_rfid_reader.py.default
          rm daemon_rfid_reader.py
          cp daemon_rfid_reader.py,kkmoonRFIDreader daemon_rfid_reader.py

Die anpasste "daemon_rfid_reader.py,kkmoonRFIDreader" müsste natürlich in den Branch mit aufgenommen werden.

closed 2019-08-31 by heikon84

switzel, 2022-02-08:
Actually, why *does* Reader have a reader attribute [(itself)](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/b3e91ae567cd3d089708f5b9d1a5f634bb41e8db/scripts/Reader.py#L35)? Shouldn't this just be removed and the extra `reader.` be dropped in general?