MiczFlor/RPi-Jukebox-RFID #552: phoniebox plays only mp3

Spuckiii, 2019-06-13:
Hello,
maybe somebody can help me. I have installed phoniebox with spotify. after some trouble to get music out off the speaker i have the next problem. Spotify links will not be played and from the locale files phoniebox plays only mp3. Anyone here who can help me?

OS Distribution
Raspbian
Description
Raspbian GNU/Linux 9.9 (stretch)
Release
9.9
Codename
stretch
Phoniebox Setup
Version
1.1.9
Edition
Plus edition (feat. Spotify integration)
Mopidy Server Status

active (running) since Thu 2019-06-13 23:40:54 CEST; 18min ago

MiczFlor, 2019-06-21:
Hi @Spuckiii 
if I remember correctly, you have had at least one Phoniebox install running perfectly. What you describe is so weird that I don't know how to investigate this over the forum. Did you get it fixed? Did a reinstall fix it? If so, please add your solution here and close the thread. 
Sorry to be of such little help.

MiczFlor, 2019-06-28:
Hi @Spuckiii 
there are two scripts in the scripts/helperscripts directory:

* [scripts/helperscripts/CreateSampleAudiofoldersStreams.sh](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/helperscripts/CreateSampleAudiofoldersStreams.sh)

This will create sample folders with different kinds of media. One folder will be created that is called `ZZZ-AudioFormatsTest` and contains a number of different audio files. Please have a look if these are not played at all, if only the mp3 file is played or if some are played and others not.

You can remove the sample folders and files with this script:

* [scripts/helperscripts/DeleteSampleAudiofoldersStreams.sh](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/master/scripts/helperscripts/DeleteSampleAudiofoldersStreams.sh)

MiczFlor, 2019-07-18:
Silence is golden? :)

closed 2019-07-18 by MiczFlor