MiczFlor/RPi-Jukebox-RFID #553: Jukebox laesst sich nicht starten

DerLange87, 2019-06-28:
Ich nutze einen Pi 2 Model B
mit einem Edimax EW-7811Un WLAN Stick

Ich habe alles installiert mit dem one-line install script.

Wenn ich nun die Jukebox oeffnen moechte wie beschrieben ueber die IP Adresse 192.168.1.199 kommt der Fehler 
'' Diese Website ist nicht erreichbar
192.168.1.199 hat die Verbindung abgelehnt. ''

Auch wenn ich versuche ueber http://localhost/ zu oeffnen kommt die selbe Fehlermeldung.

Ich bin leider sehr neu im Bereich programmierung und dies ist mein erstes Projekt mit dem Pi.
Wuerde mich ueber Hilfe sehr freuen.

gruesse

MiczFlor, 2019-06-29:
Hallo @DerLange87
Um das Problem zu verstehen:

1. Du hast den Pi mit monitor, Keyboard, Maus verbunden, da den Browser aufgemacht un localhost zeigt nichts an. 

2. Oder versuchst du die Phoniebox über das wlan vom Handy zu erreichen?

Probier auf alle Fälle mal Version 1 und schreibe was passiert. 

Und woher hast du die IP Adresse 192.168.1.199!

DerLange87, 2019-06-29:
Guten morgen. 

Oh ja, das hätte ich noch dazu sagen müssen. 
Also ich habe den Pi per HDMI am Monitor angeschlossen und Maus und Tastatur. Die IP und Adresse habe ich im Browser des Pi eingegeben. Die IP habe ich in der Installation festgelegt weil ich sie als statische Adresse nutzen möchte. 

Wenn ich vom Handy, welches im selber WLAN ist die IP Adresse aufrufen will kommt "Verbindung fehlgeschlagen"

m4w, 2019-06-29:
Ich habe das selbe Problem. Meiner Meinung nach scheint das Script nicht alle nötigen dependencies zu installieren:
```
Note, selecting 'linux-headers-4.9.0-6-all' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-rpi' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-all-armhf' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-rpi2' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-common' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-common-rt' for regex 'linux-headers-4.9'
Note, selecting 'php7.0-thrift' for regex 'php7.0'
Note, selecting 'php7.0-remctl' for regex 'php7.0'
Note, selecting 'php7.0-libsodium' for regex 'php7.0'
Note, selecting 'php7.0-common' for regex 'php7.0'
Note, selecting 'php7.0-curl' for regex 'php7.0'
Note, selecting 'php-libsodium' instead of 'php7.0-libsodium'
Note, selecting 'php-remctl' instead of 'php7.0-remctl'
Package php7.0-common is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

W: --force-yes is deprecated, use one of the options starting with --allow instead.
E: Package 'php7.0-common' has no installation candidate
E: Unable to locate package php7.0-cgi
E: Couldn't find any package by glob 'php7.0-cgi'
E: Couldn't find any package by regex 'php7.0-cgi'
E: Unable to locate package php7.0-fpm
E: Couldn't find any package by glob 'php7.0-fpm'
E: Couldn't find any package by regex 'php7.0-fpm'
```
Diese Hinweise wärend der Installation sprechen dafür. Nach dem ersten Versuch war zum Beispiel auch der Webserver lighttpd nicht installiert. Den hatte ich manuel nachinstalliert. Hat aber auch nichts gebracht. 
Beim Restart des webservers gibt es auch folgende Meldung `install: cannot create directory '/var/cache/lighttpd': Permission denied` 

Wahrscheinlich ist das Installscript nicht mehr funktionsfähig. Leider verstehe ich zu wenig davon, um zu wissen, wo ich Hand anlegen kann, um das zu beheben.

m4w, 2019-06-29:
Der vollständigkeithalber noch einmal der gesamte Log:
And a short introduction for the rest of the world: There is a problem to start the jukebox. It is not working out of the box after a fresh install. Also there are many warnings and errors during the install process:

```#####################################################
#
# START INSTALLATION
#
# Good news: you completed the input.
# Let the install begin.
#
# Get yourself a cup of something. The install takes
# between 15 minutes to half an hour, depending on
# your Raspberry Pi and Internet connectivity.
#
# You will be prompted later to complete the installation.

Do you want to start the installation? [Y/n] y
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Ign:2 http://apt.mopidy.com stretch InRelease
Hit:3 http://archive.raspberrypi.org/debian buster InRelease
Hit:4 http://apt.mopidy.com stretch Release
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'linux-headers-4.9.0-6-all' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-rpi' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-all-armhf' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-rpi2' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-common' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-common-rt' for regex 'linux-headers-4.9'
Note, selecting 'php7.0-thrift' for regex 'php7.0'
Note, selecting 'php7.0-remctl' for regex 'php7.0'
Note, selecting 'php7.0-libsodium' for regex 'php7.0'
Note, selecting 'php7.0-common' for regex 'php7.0'
Note, selecting 'php7.0-curl' for regex 'php7.0'
Note, selecting 'php-libsodium' instead of 'php7.0-libsodium'
Note, selecting 'php-remctl' instead of 'php7.0-remctl'
Package php7.0-common is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

W: --force-yes is deprecated, use one of the options starting with --allow instead.
E: Package 'php7.0-common' has no installation candidate
E: Unable to locate package php7.0-cgi
E: Couldn't find any package by glob 'php7.0-cgi'
E: Couldn't find any package by regex 'php7.0-cgi'
E: Unable to locate package php7.0-fpm
E: Couldn't find any package by glob 'php7.0-fpm'
E: Couldn't find any package by regex 'php7.0-fpm'
OK
Ign:1 http://apt.mopidy.com stretch InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:3 http://apt.mopidy.com stretch Release
Hit:4 http://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
mopidy is already the newest version (2.2.2-1).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
W: --force-yes is deprecated, use one of the options starting with --allow instead.
Reading package lists... Done
Building dependency tree
Reading state information... Done
python-cffi is already the newest version (1.12.2-1).
python-ply is already the newest version (3.11-3).
python-pycparser is already the newest version (2.19-1).
python-spotify is already the newest version (2.0.5-0mopidy1).
libspotify12 is already the newest version (12.1.103-0mopidy1).
[... 102 lines cut ...]
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_CTYPE = "de_DE.UTF-8",
	LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
mv: cannot stat '/home/pi/BACKUP/shared/shortcuts/*': No such file or directory

#
# INSTALLATION FINISHED
#
#####################################################

If you are using an USB RFID reader, connect it to your RPi.
(In case your RFID reader required soldering, consult the manual.)
Have you connected your USB Reader? [Y/n] n

DONE. Let the sounds begin.
Find more information and documentation on the github account:
https://github.com/MiczFlor/RPi-Jukebox-RFID/tree/master/docs/```

DerLange87, 2019-06-29:
Stimmt. Einige solche Fehler hatte ich auch bei der Installation. Dachte mir aber nichts dabei. Also Installation wiederholen? Woran kann es liegen das die Fehler auftreten? Kann ich was dagegen tun? Und kann ich das one-line install Script einfach noch mal drüber laufen lassen oder muss ich das alte erst deinstallieren?

m4w, 2019-06-29:
Bei mir hat das leider nicht funktioniert. Ich wollte einmal eine händische Installation probieren, aber dabei jedes Paket einzelnd. Dann kann man vielleicht noch etwas her raus finden.

MiczFlor, 2019-06-29:
Oh. Wow. Ich werde die Installation selber noch mal probieren (müssen).

Scheinbar geht das vielen so. Und ich habe davon noch nie gehört. Vielleicht hat sich was im Raspbian geändert. 

Welche Version habt ihr installiert?

DerLange87, 2019-06-29:
Ähm das ist ein eine gute Frage. Glaube die, die empfohlen wurde da darauf getestet wurde. Wie bekomm ich das heraus?

m4w, 2019-06-29:
Moin, ich habe die neuste Version Raspbian Buster Lite. Kann aber auch die ältere Variante mal ausprobieren, sobald ich mal wieder Zeit habe.

MiczFlor, 2019-07-01:
Ich bin noch nicht zum Testen bekommen. In Wiki unter „install“ gibt es einen Link zur Raspbian version die auf alle Fälle getestet ist:
IMPORTANT: if you want to be sure that you have the same system running that this documentation was written for, you need to use the stretch distribution which you can download here: 2018-03-13-raspbian-stretch.zip.

DerLange87, 2019-07-01:
Genau diese Version habe ich ja genommen

m4w, 2019-07-01:
Ich habe nun die Version 2018-03-14-raspbian-stretch-lite genommen. Es klappt soweit, dass der Webserver richtig startet. Also lag es bei mir an der falschen Version. Vielen Dank

DerLange87, 2019-07-01:
Ich habe auch noch mal diese runter geladen und das System neu gemacht. Bei der Installation sind wieder rote Fehler durch gelaufen. Diesmal erreiche ich aber die Box über den Browser des Pi wenn ich localhost eingebe. Über ip und Handy nicht erreichbar

DerLange87, 2019-07-04:
Moin. Ich habe mich heute wieder an mein Raspberry gesetzt und hab weiter probiert. Nun erreiche ich die phoniebox über die IP. Soweit alles gut. 

Allerdings macht jetzt der RFID reader Probleme. 
Wenn ich eine Karte registrieren möchte wird die ID nicht in der Oberfläche angezeigt.
Versuche ich es woanders klappt es. Also beispielsweise im Browser.
In den Einstellungen steht er auf deaktiviert. Stelle ich auf aktiviert, bleibt das nur solange bis ich aktualisiere. Ich habe einen anderen USB Port versucht, da hat es ein mal geklappt. Seit dem nie wieder. Was kann ich machen?

DerLange87, 2019-07-06:
Ich habe nun heraus gefunden, wenn ich in das Feld wo die Karten ID eingetragen wird vorher hinein klicke, die Karte dann drüber halte, und das ein paar mal mache die ID dort angezeigt wird.

kann mir bitte jemand helfen das es klappt ohne das ich dort hinein klicken muss?

liebe Grüße

MiczFlor, 2019-07-07:
Hallo @DerLange87 
welchen Browser in welchem Betriebssystem nutzt du? Hast du mal einen anderen Browser versucht? Ich teste immer nur auf Chrome und Firefox (beides auf Linux), weshalb es sein kann, dass sich da ein Fehler in einem anderen Browser zeigt. Bis jetzt hatte ich von dem Problem aber noch nicht gehört.
Beste Grüße, Micz

closed 2019-07-12 by MiczFlor