MiczFlor/RPi-Jukebox-RFID #554: Speed up web interface by PHP caching

RogoTe, 2019-07-02:
Tip: one can speed up the web interface of the phoniebox significantly by extending the PHP caching.
Assuming the default Raspbian Stretch lite installation PHPs inbuilt opcache is already enabled.  This can be speed up by disabling rechecking files in PHP configuration.
`opcache.validate_timestamps = 0`
Otherwise the PHP process has to access the file system every x seconds to ensure the PHP file has not changed.

Further speed up can be reached by installing PHP module APCu. This is an additional caching mechanism. While opcache caches the compiled .php files bytecode, APCu tries to cache (and even prefetch) database queries, file system accesses and calculations this bytecode does.

Assuming default Raspbian Stretch lite installation run this commands via SSH

    sudo apt-get install php-apcu
    
    cd /etc/php/7.0/fpm/conf.d/
    
    sudo touch 999-RPi-Jukebox-RFID.ini
    sudo sh -c 'echo "opcache.validate_timestamps = 0" >> 999-RPi-Jukebox-RFID.ini'
    sudo sh -c 'echo "apc.preload_path=\"/home/pi/RPi-Jukebox-RFID/htdocs\"" >> 999-RPi-Jukebox-RFID.ini'
    
    sudo service php7.0-fpm restart

Please be aware that this leads to heavy caching (and significant speed up); to see changes in PHP code you made you will have to restart the PHP-FPM.

RogoTe, 2019-07-02:
@MiczFlor In case you like this, I could compile a simple pull request.

MiczFlor, 2019-07-07:
Hi @RogoTe,
could you prep a pull request for the develop branch? I would like to try this.
Thanks for the contribution!
All the best, micz

closed 2020-01-18 by MiczFlor