MiczFlor/RPi-Jukebox-RFID #556: SPI update for rc522

SpaceC, 2019-07-08:
Is it possible to include an update of:

https://github.com/ondryaso/pi-rc522/blob/master/pirc522/rfid.py

Background: 
My distance card to reader is about some 10 mm this causes many "new card" events.
This leads to songs starting over and over again. 
This can be avoided using the setting. 
"Nur Systembefhle mehrfach ausführen"

But now the following issue ocours.
After sometime, no new card is recognized. WiFi is still there, playback via webfrontend is working, Buttons still working.
But now other song with RFID.

Therefore I started to setup my Box again and during installation I had some time to read....
Found the comment in the link and double checked the datasheet and yes it makes sense.

I stoped the new installation and switched back to the old ones. 
Exchanged the file (changed also the comments from C style to pyhton style) and I had no issue again now after 2h testing.

Thanks for that great Project!!

MiczFlor, 2019-08-16:
Hi @SpaceC 
thanks for the pointer. I am not sure I understand what you suggest to be done. Could you rephrase it for me?

Menelmacar, 2020-01-04:
I have the same issue (with rc 522 RFID reader, stops working after 30s - 5 min)
It'S wired as described in the manual. 
Version 2.0 rc4
Edition Classic edition (barebones)
 on RPI 3b Buster fresh install.
After install I replaced Reader.py with experimental as the manual says.
The Box works totally fine, but after a short amount of usage no new cards are recognised. This happens only if a card stays on the Reader. If it is only swiped one over the issue will not appear.

After a systemctl restart phoniebox-rfid-reader.service it will work again for some time.

To debug I started the daemon_rfid_reader.py manually. 
In the console I can see an update of Card detected like two per second until it suddenly stops updating until I stop manually...
Ouptput:
---------------------------------------------------------------
...
Card detected.
Card detected.
^CTraceback (most recent call last):
  File "/home/pi/RPi-Jukebox-RFID/scripts/daemon_rfid_reader.py", line 22, in <module>
    cardid = reader.reader.readCard()
  File "/home/pi/RPi-Jukebox-RFID/scripts/Reader.py", line 57, in readCard
    self.device.wait_for_tag()
  File "/home/pi/RPi-Jukebox-RFID/scripts/pirc522/rfid.py", line 401, in wait_for_tag
    waiting = not self.irq.wait(0.2)
  File "/usr/lib/python2.7/threading.py", line 614, in wait
    self.__cond.wait(timeout)
  File "/usr/lib/python2.7/threading.py", line 359, in wait
    _sleep(delay)
---------------------------------------------------------------------------

Maybe the issue is with the "waiting = not self.irq.wait(0.2)" ? 
I am no coder and have no clue how to debug it further..

s-martin, 2020-03-07:
@ZyanKLee Should we include https://github.com/ondryaso/pi-rc522 as dependency and remove the sources from our repo?

ZyanKLee, 2020-03-07:
@s-martin I'd say this is always preferable as long as there aren't any changes in our local version that were never mainlined into the upstream project.

s-martin, 2020-03-07:
I’ll check, if we have any changes on our repo.

s-martin, 2020-03-07:
If have just seen that we already install pi-rc522 via [requirements.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/requirements.txt)

ZyanKLee, 2020-03-07:
in that case I will remove `scripts/pirc522` and see what breaks :-D

ZyanKLee, 2020-03-07:
please test if this still solves your issue @Menelmacar and @SpaceC

closed 2020-03-19 by MiczFlor