MiczFlor/RPi-Jukebox-RFID #557: Volume control not working with Justboom Amp Hat and GPIO buttons enabled

FreeMinded, 2019-07-10:
I'm building an RPi-Jukebox with Raspberry Pi 3, an JustBoom Amp Hat and the recommended Neuftech USB RFID Reader. Once I switched to the flexible GPIO pin layout as recommended in the wiki I got the sound working. 
But as soon as I activate the GPIO buttons the volume control doesn't work anymore (Vol Down/Up, Mute). I can see the volume "knob" move in the web interface, but it doesn't change anything. It's also not possible to set the volume from the web interface. The value changes, but not on the output side. Power, Skip, etc buttons do actually work.
When I disable the GPIO buttons I can control the volume through the web interface, no problem.
With Alsamixer I see the "digital" level move as I change the volume in the web interface. And I can change the volume in Alsamixer and see the change in the web interface, too.
With GPIO buttons enabled nothing moves in Alsamixer, and trying to change the volume from Alsamixer doesn't change anything neither.

I tried to disabled the volume control pins in the GPIO-buttons script among other things. But it still does not work.

Has anyone an idea on how to fix this?

FreeMinded, 2019-07-13:
I just figured that the issue is actually the shutdown function. If I comment that pin out (GPIO3), volume works as expected. According to the hardware pinout overview [1] this should work in conjunction with I2C devices.
Any idea on this?

1: https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hardware-Pinout-Overview

Lavaeolous, 2019-08-09:
You use GPIO 3 for your Amp Hat as well as for the Shutdown. I think you need to re-assign the shutdown to a different pin

closed 2020-01-18 by MiczFlor

FreeMinded, 2020-01-20:
The solution is to use a different pin for shutdown (I use GPIO 7) but also wiring GPIO 3 on the same button to be able to wake up the Pi. Thanks!