MiczFlor/RPi-Jukebox-RFID #558: RPi-Jukebox installer not comatible with raspbian buster

ckuetbach, 2019-07-11:
The installscript shows lots of errors.

```
Do you want to start the installation? [Y/n]
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; No such device.
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'linux-headers-4.9.0-6-all' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-rpi' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-all-armhf' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-rpi2' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-common' for regex 'linux-headers-4.9'
Note, selecting 'linux-headers-4.9.0-6-common-rt' for regex 'linux-headers-4.9'
Note, selecting 'php7.0-thrift' for regex 'php7.0'
Note, selecting 'php7.0-remctl' for regex 'php7.0'
Note, selecting 'php7.0-common' for regex 'php7.0'
Note, selecting 'php7.0-curl' for regex 'php7.0'
Note, selecting 'php-remctl' instead of 'php7.0-remctl'
Package php7.0-common is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

W: --force-yes is deprecated, use one of the options starting with --allow instead.
E: Package 'php7.0-common' has no installation candidate
E: Unable to locate package php7.0-cgi
E: Couldn't find any package by glob 'php7.0-cgi'
E: Couldn't find any package by regex 'php7.0-cgi'
E: Unable to locate package php7.0-fpm
E: Couldn't find any package by glob 'php7.0-fpm'
E: Couldn't find any package by regex 'php7.0-fpm'
Cloning into 'RPi-Jukebox-RFID'...
remote: Enumerating objects: 4843, done.
remote: Total 4843 (delta 0), reused 0 (delta 0), pack-reused 4843
Receiving objects: 100% (4843/4843), 11.15 MiB | 1.55 MiB/s, done.
Resolving deltas: 100% (3218/3218), done.
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
sudo: pip: command not found
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; No such device.
Added user pi.
cp: cannot create regular file '/etc/php/7.0/fpm/php.ini': No such file or directory
chown: cannot access '/etc/php/7.0/fpm/php.ini': No such file or directory
chmod: cannot access '/etc/php/7.0/fpm/php.ini': No such file or directory
sudo: lighttpd-enable-mod: command not found
sudo: lighttpd-enable-mod: command not found
Failed to restart lighttpd.service: Unit lighttpd.service not found.
Failed to restart php7.0-fpm.service: Unit php7.0-fpm.service not found.
### Deleting older versions of service daemons. This might throw errors, ignore them
Failed to disable unit: Unit file idle-watchdog.service does not exist.
Failed to disable unit: Unit file rfid-reader.service does not exist.
Failed to disable unit: Unit file startup-sound.service does not exist.
Failed to disable unit: Unit file gpio-buttons.service does not exist.
rm: cannot remove '/etc/systemd/system/rfid-reader.service': No such file or directory
rm: cannot remove '/etc/systemd/system/startup-sound.service': No such file or directory
rm: cannot remove '/etc/systemd/system/gpio-buttons.service': No such file or directory
rm: cannot remove '/etc/systemd/system/idle-watchdog.service': No such file or directory
```

Seems to be related to https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/553

MiczFlor, 2019-07-12:
Hi @ckuetbach 
that is true. Following your post I am adding these lines to the install instructions (and probably you want to give this a go? :)

*Written for an tested on Raspbian GNU/Linux 9.4 (stretch) Kernel version: 4.14*

**IMPORTANT**: the install script works and has been tested with the following stretch distribution which you can download here: [2018-03-13-raspbian-stretch.zip](https://downloads.raspberrypi.org/raspbian/images/raspbian-2018-03-14/2018-03-13-raspbian-stretch.zip).

If you want to modify the install script to run on `buster`, commit your work (under the filename: `buster-install-default.sh`) to the develop branch and create a pull request.

ckuetbach, 2019-07-12:
Looks good to me. 

Right now, I am _trying_ to create a `buster-install-default.sh`.

ckuetbach, 2019-07-15:
I just started: https://github.com/ckuetbach/RPi-Jukebox-RFID/tree/buster-install and it looks somewhat promising. I am n ot sure, how I can make the script _not_ load from you github-repo while installing. Bu as long if I create new files and not edit existing ones, I getr further :)

MiczFlor, 2019-07-18:
Hi @ckuetbach 
Thanks! I will be off until the 5th of August and then make the time to bust into buster. 
You could create a pull request for your work or I will copy it and then work from there. Looking forward to see how far you‘ll be. The main issues will be the apt-get install line and possibly the config files for webserver and mpd. 
Good luck :)
All the best, micz

ckuetbach, 2019-07-18:
The apt-get part wasn't hard, but the strech sudo file seems to be incompatible for buster. Now I can't sudo anymore ;) 

Why do you include the sudoer config?

ckuetbach, 2019-07-20:
Well... I should not let git change the lineendigs to windows and use this file als sudoers ;( My fault.

ckuetbach, 2019-07-22:
I think I got it; But I need to make some tests with my RFID reader and the latest release Buster-Image.

I forked the master branch and now I am behind development-branch. Should I merge your development-branch into my buster-feature?

MiczFlor, 2019-07-23:
Hi @ckuetbach 
Thanks! If you create a pull request to master, i can change the base to develop. So no need to make it more complicated for you. 
Kudos to open a new chapter :)
All the best, micz

ckuetbach, 2019-07-31:
I still have one error:

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting evdev==0.7.0 (from -r requirements.txt (line 7))
  Downloading https://files.pythonhosted.org/packages/67/15/eac376f3e1fc1960a54439c21459b2582e68340001aff83b4ace9e5bd110/evdev-0.7.0.tar.gz
Collecting spi-py from git+git://github.com/lthiery/SPI-Py.git#egg=spi-py (from -r requirements.txt (line 8))
  Cloning git://github.com/lthiery/SPI-Py.git to /tmp/pip-install-owNU8Z/spi-py
Collecting youtube_dl (from -r requirements.txt (line 9))
  Downloading https://files.pythonhosted.org/packages/b1/ec/fe552181d6bd05a9e5b6b51f6f7ea4fed9f121ce595d788217e59318e47c/youtube_dl-2019.7.30-py2.py3-none-any.whl (1.8MB)
    100% |████████████████████████████████| 1.8MB 22kB/s
Collecting pyserial (from -r requirements.txt (line 10))
  Downloading https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl (193kB)
    100% |████████████████████████████████| 194kB 149kB/s
Collecting spidev (from -r requirements.txt (line 11))
  Downloading https://files.pythonhosted.org/packages/fb/14/4c2e1640f0cb04862c76d9d76ed7c945b0f67876e503ac02f7f675fe86a0/spidev-3.4.tar.gz
Requirement already satisfied: RPi.GPIO in /usr/lib/python2.7/dist-packages (from -r requirements.txt (line 12)) (0.6.5)
Collecting pi-rc522 (from -r requirements.txt (line 13))
  Downloading https://files.pythonhosted.org/packages/84/f2/e3d02257949e9caa9bff26044e0185e743ff45d3e0e099a93880e10bf718/pi-rc522-2.2.1.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-owNU8Z/pi-rc522/setup.py", line 11, in <module>
        from pirc522 import __version__  # flake8: noqa
      File "pirc522/__init__.py", line 4, in <module>
        from .rfid import RFID
      File "pirc522/rfid.py", line 2, in <module>
        import spidev
    ImportError: No module named spidev

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-owNU8Z/pi-rc522/
```

This seems to be my last error. Any ideas?

AdmiralVS, 2019-08-15:
Greetings Gents! Is there any update on this thread (installer for Buster)? I'm about to build my fourth box (for my three kids and my goddaughter), and this will be the first on Buster, so: can I try the current installer and help if needed?

MiczFlor, 2019-08-16:
Hi @AdmiralVS hi @ckuetbach 
I added @ckuetbach first steps to the `develop` branch:

https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/develop/scripts/installscripts/buster-install-default.sh

@AdmiralVS please take it and run with it :) I will also work on the weekend on a buster install, so we might both want to create pull requests frequently... :)

And congrats to the fourth Phoniebox. Have you posted any pictures already to:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/5

All the best, micz

MiczFlor, 2019-08-16:
I will work with the raspbian lite image found here:
https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-07-12/2019-07-10-raspbian-buster-lite.zip

AdmiralVS, 2019-08-16:
Pictures! Of course! I've added them for two I have in house.

I'm basing my install on the Raspbian Lite you get from NOOBS; that's probably the same as yours.

The part about the "frequent pull requests" implies I know how to speak git and github. Which I don't really. I did manage to clone the repository and switch to the develop branch. And now:
- just checking: there are mentions of Buster files in the installer, e.g. "lighttpd.conf.buster-default.sample" but they aren't there, yet. Right?
- I'm installing packages and dependencies via apt where possible as opposed to via sudo pip install. Is there a problem with that approach as it seems somewhat cleaner from my perspective. But then: I'm not a huge Python guy either.
- I'd be happy to contribute these little changes already. Any hints how to do that gratefully received.

And, by the way: Thanks for this great project!

MiczFlor, 2019-08-17:
Hi @AdmiralVS 
I guess we are both working on the same "latest" Raspbian.
Have you started? Because I could try to push the conf files to the develop branch.
Regarding the install process via pip or sudo, makes little difference at this stage and I can clean that up once we have somethng :)

Awesome design work on your phonieboxes, btw!

AdmiralVS, 2019-08-17:
Hi @MiczFlor 
I have barely started. I believe I circumvented the last error reported by @ckuetbach (about missing spidev while building pi-rc522) by installing spidev from apt. So software wise I believe everything is installed. If you already have Buster versions of some of the config files I'd be happy to plug them in.
What is the preferred way to share my modifications?

The design of the boxes is courtesy of my wife. I'll relay the praise. :-)

MiczFlor, 2019-08-18:
Hi @ckuetbach @AdmiralVS 
I think I got something close enough to working:
~~~
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh
~~~
It's also in the wiki, so I will close this thread and start a new one with into about the buster install at the top.

closed 2019-08-18 by MiczFlor

ckuetbach, 2019-08-19:
I hope I was helpful. This is a great project and my children loves their box.

MiczFlor, 2019-08-20:
Hi @ckuetbach 
don't forget to share pictures as inspiration - and potentially for the 2020 Phoniebox calendar coming up in December :)
Also, if you find the time, please have a look at the install script, if it works for you. Sometimes the devil is in the detail...