MiczFlor/RPi-Jukebox-RFID #559: RFID löst keine Aktion aus

EliEquitania, 2019-08-03:
Folgende Situation:
Raspberry B+ aufgesetzt wie in der Beschreibung https://youtu.be/spHDGSxckmw
RFID-Reader wie dort beschreiben besorgt:  https://amzn.to/2OMMRHX

Wenn ich die Web-App starte, ist der Reader aktiv. Wenn ich in der Kommandozeile eingebe
**./rfid_trigger_play.sh -i=**
und den Chip über den Reader ziehe, werden die zugeordneten Ordner abgespielt oder die Chipnummer in die Web-App eingetragen. 

Es gab mal ein ähnliches Issue "https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/231". Die Einträge endeten mit verschiedenen Versionen der reader.py. Die habe ich alle ausprobiert (originale Reader.py, Reader.py.experimental und die Reader.py.kkmoonRFIDreader), keine hat die Lösung gebracht. 

Jetzt weiß ich nicht mehr weiter.  Die Idee von der Phoniebox ist so genial, weil sie auch tausendmal flexibler als das Original ist. Kann mir jemand helfen?

EliEquitania, 2019-08-04:
Ich kaufe auch gerne einen anderen RFID-Leser, wenn mir einer sagt, welcher funktioniert.

EliEquitania, 2019-08-05:
Ich werde erstmal die Lösung aus dem Issue https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/551 versuchen.

Lavaeolous, 2019-08-05:
Hey, ich hab den gleichen Reader verbaut. Der funktioniert bei mir ohne Probleme mit dem One-Line Install.
Hast du [Raspian Stretch](http://downloads.raspberrypi.org/raspbian/images/raspbian-2019-04-09/) oder die neuste Version [Buster](https://www.raspberrypi.org/downloads/raspbian/)?

Falls du es mit Buster versucht hast, setz den Pi mal mit Stretch neu auf.

EliEquitania, 2019-08-05:
Leute ist sag's ja, meistens sitzt der Fehler vor der Tastatur. So auch hier. Ich habe einfach das richtige USB Device eingegeben und schon gings. Zu meiner Entschuldigung muss ich gestehen, dass ich auch eine Maus uns eine Tastatur am PI angeschlossen habe und wahrscheinlich deswegen auf Irrwegen war.

Meine USB_Devices hat das  Installationsscript ausgegeben (s. Foto). Da kommt doch keiner drauf, dass der Reader das Device  HXGCoLtd ist!. Peiblich, peinlich. Aber das gibt mir jetzt die Gelegenheit zu sagen, wie geil das Projekt ist! Ungeahnte Möglichkeiten biten sich damit, einfach toll. Danke für die Arbeit! 
[image]

closed 2019-08-05 by EliEquitania

MiczFlor, 2019-08-07:
Hi @EliEquitania 
Danke für die Rückmeldung. Oft bleiben Tickets offen und ich trau mich nicht zu schließen :)
Ich komme jetzt erst zum antworten. Zwischendrin hatte ich schon gegrübelt warum der daemon nicht läuft. Hat sich jetzt geklärt. Danke auch für die Lorbeeren und ich bin gespannt auf Fotos!!!
Beste Grüße, Micz