MiczFlor/RPi-Jukebox-RFID #561: Phoniebox does not play any sounds

Sm4rtCookie, 2019-08-08:
Hi everyone,

Thanks for this great application and great documentation.

I personally though have a problem getting the Phoniebox playing any sounds.

I have a Raspberry Pi 3 Model B and I first installed the "2018-03-13-raspbian-stretch"
and afterwards "Phoniebox +Spotify Edition" via One Line Command Installation.
Everything looked good so far. 

- I am able to access the web frontend
- RFID reader works fine
- Cards can be assigned to MP3s and Spotify playlists

The only problem is, that the phoniebox is not playing anything. When manually pressing on play in the front end, the page reloads but nothing happens. THe play button remains the play button and doesnt change to Pause button. No sound comes out of the speakers (VLC player in raspbian plays sounds - so no hardware issue here).

I am completely new to Raspberry and Linux so I was not really able to deep dive into the settings and logs. But what I was able to find in one of the logs i found is the following:

_Line 2539: Aug  5 20:34:43 raspberrypi mopidy[437]: ERROR    GStreamer error: gst-core-error-quark: Your GStreamer installation is missing a plug-in. (12) 

   Line 2580: Aug  5 20:34:56 raspberrypi mopidy[437]: WARNING  GStreamer warning: gst-stream-error-quark: No decoder available for type 'text/html'. (6)_ 

I dont know if this of any help...

Another thing I noticed which looks strange is the amount of entries of new MPD connections. I have four entries per second like this:
Aug  6 21:00:41 raspberrypi mopidy[486]: INFO     New MPD connection from [::ffff:127.0.0.1]:6600

I installed the raspbian and Phoniebox(spotify) three times now on two different sd cards and still the same issue.

Do you have an idea what I might have done incorrectly?

Thanks in advance for your help!
SmartCookie

Sm4rtCookie, 2019-08-08:
Spotify Iris Plays Sound - btw.

MiczFlor, 2019-08-08:
Hi @Sm4rtCookie 
before I put my thinking cap on, did you read the trouble shooting section?
https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working

Sm4rtCookie, 2019-08-08:
Hi @MiczFlor ,
- Version: 1.1.9
- Internal Soundcard (=PCM)
- Raspberry plays sound via VLC and also the SpotifyIris player plays sounds
I also tried to find any other solution in the issues here - but no luck so far.

Another observation: After a reboot the volume is set to 0% and I cannot set it higher.

When trying to reload the data with mpc update, the following message comes:
pi@raspberrypi:~ $ mpc update
#mpd error: Connection closed by the server

ddbkva, 2019-08-08:
do you use a powerbank ?

Sm4rtCookie, 2019-08-08:
> do you use a powerbank ?

No, but I plan to use one once it is working.

Sm4rtCookie, 2019-08-10:
Hey everyone,
I did another installation with the same result. 
Afterwards I went through some parts of the manual installation instructions and noticed that the etc/mpd.conf file did not contain the according entries. I had to change the values manually. Unfortunately this still didn't help...
Today in the evening I will run the Analytics_AfterInstallScript.sh script

MiczFlor, 2019-08-10:
Try ‚Speaker‘ instead of ‚PCM‘ when installing. 
Also have a read here:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/519
I can‘t think of anything else at tje moment - miles away from a phoniebox :/

Sm4rtCookie, 2019-08-12:
I accidentally manually installed the version without spotify. here everything works fine. AFter pressing the play button (or scanning a card via RFID) the mp3 file is being played and the track status is shown (in three second steps). This exactly what is not working with the spotify version (not even playing mp3s).
I think this has something to do with mpd and mopidy. 
Do I need to install and run MPD even if I want to use mopidy for spotify (I hope this question makes sense)?

MiczFlor, 2019-08-13:
I have no Spotify myself. But I recall from  another conversation that mpd and mopidy require that the conf is set in more than one place. Possibly you can find more about this when searching the web.

Sm4rtCookie, 2019-08-15:
I dont know what exactly the issue was but now it is working. I exactly wrote down what my steps were - maybe it helps someone else:

	- Install raspbian stretch -> 2018-03-13-raspbian-stretch.zip
	- SSH and wpa_supplicant.conf copied to boot
	- Sudo raspi-config
		○ Boot options changed
		○ Keyboard options adjusted
	- sudo apt-get update
	- Sudo apt-get upgrade
	- sudo systemctl enable mopidy
	- One line command spotify
		○ cd; rm stretch-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/master/scripts/installscripts/stretch-install-spotify.sh; chmod +x stretch-install-spotify.sh; ./stretch-install-spotify.sh
	- No WIFI configuration
	- reboot
	- cd /home/pi/RPi-Jukebox-RFID/scripts/helperscripts/; ./CreateSampleAudiofoldersStreams.sh
	- sudo systemctl disable mpd
	- sudo systemctl enable mopidy

closed 2019-08-15 by Sm4rtCookie

MiczFlor, 2019-08-16:
Hi @Sm4rtCookie ,
thanks for sharing the steps. Seems that you had your very own "ghost in the machine" haunting your Pi.
All the best, micz

j-gimbel, 2020-09-11:
Hello, I seem to have the same issue. 
I cannot play music via webinterface or with a RFID card.

raspian buster, 
PRI-Jukebox version 2.0 - ed888da - master . , 
with Spotify.

- I hear sound when the system boots.
- Spotify Iris plays sounds, 
- command line mpg123 also plays sounds. 
- Card gets detected and cofigured ( I created a folder "test" with only one file "test.mp3" and connected it to the card.

But when I swipe the card: nothing happens (cards get recognized, but no sound, no play/pause button change.
When click the play button nothing happens.

In the command line the command "mpc playlist test" lists the playlist. 
But "mpc playlist" shows an empty list.
So it seems that somehow mpd does not get the playlist. 

But when I manually add the file via "mpc add" I can start the music over the webinterface (not with the card).
Any ideas ?
Maybe the audio files need certain ownerships (I gave the 777 permissions to rule out access issues) .

s-martin, 2020-09-12:
What’s the output of `journalctl -u mopidy`?