MiczFlor/RPi-Jukebox-RFID #562: RFID Extension - Barcode Scanner / Webcam for cheaper "card production"

sandreas, 2019-08-12:
Hi,

i  like the idea, that rfid chips are cheap and easy to use. But would it be possible to also use a barcode scanner to trigger track changes?

This should be a similar technology and since a barcode is something that only needs to be printed, it would be easier to produce... perhaps also in combination with a barcode generator in the web interface to have unique id's...

As i own an old webcam, it would be also great to combine this with a webcam barcode reader - like this: https://www.raspberrypi.org/forums/viewtopic.php?t=225499

What do you think?

dnmtch, 2019-08-15:
I would recommend working (test) with a barcode scanner. (It can also be in continuous operation if necessary). The barcode scanner should behave exactly like the RFID scanner? As soon as it scans a valid barcode, it outputs the value (1234567890) as a HID device (like a keyboard).

(disclaimer: this is not a technically sound statement. Children should not handle laser light under any circumstances).

[image]

MiczFlor, 2019-08-16:
hi @sandreas @dnmtch 
true: a USB Barcode Scanner is simple to Install. actually the Same as a RFID Scanner. 
I did some tests for a digital Archive of my parents CD collection where you would use the Barcode on the actual CD to start playout.
need to write up a howto.
the real task would be to add automated generation of the Barcodes in the web App!

dnmtch, 2019-08-17:
> the real task would be to add automated generation of the Barcodes in the web App!

That's not necessary. There are enough (simpler) other ways to generate the barcodes.

sandreas, 2019-08-18:
https://lindell.me/JsBarcode/

MiczFlor, 2019-08-18:
Hi @sandreas 
very true, there are many out there. But I like the idea of this workflow:

1. upload files and create new folder
2. go to the home screen and right click to download the matching barcode
3. print

dnmtch, 2019-08-18:
The combination of RFID (10 numbers) and barcode is not trivial from my point of view. The standard "2of5" supported by many readers should be able to be used. 
Nevertheless: A simple manual or a link to an online tool should suffice by far.
Most people will want to customize their barcode images anyway.
[image]

https://github.com/fabioboris/php-interleaved-2-of-5

dnmtch, 2019-09-05:
I have good news for you. I got the radio working right away with a barcode scanner (NETUM NT-1698W) for 12$. If there is interest, I can create a tutorial in the wiki.

If I create barcode cards, I can also publish the template.

sandreas, 2019-09-06:
@dnmtch: Thank you - a short tutorial would be very nice.

MiczFlor, 2019-09-29:
Hi @dnmtch 
yes, please add a page to the wiki! I haven't found the time to dig into this deeper. But from my first steps, it was really mainly an issue of identifying the barcode scanner instead of the RFID reader during the install?

s-martin, 2023-12-28:
Barcode scanner is essentially a keyboard input. Could be easy to adapt in future3 once PR #1943 is merged.

sandreas, 2024-01-02:
I thought of [something like this](https://www.amazon.de/exec/obidos/ASIN/B081JPN14M/wwwgeschenke-inspiration-21) together with a human body motion sensor to activate it, as soon as you put your hand with a card near the scanner.

Alternatively you could build a little wooden box around the scanner module and everytime you put a plastic card in, the activator button gets pressed or just expose the button (children should know how to use it)

Everything you would need to create cards would be a printer... no RFID needed any more.

Here is a youtube video:

https://www.youtube.com/watch?v=xX_GDkPJ6x0

s-martin, 2024-05-10:
> Barcode scanner is essentially a keyboard input. Could be easy to adapt in future3 once PR #1943 is merged.

As the evdev feature is now available in V3 it should be possible to add a barcode scanner (which is a keyboard input).

AlvinSchiller, 2024-05-10:
As long as the barcode scanner is integrated as a keyboard this should be possible right now for V2 and V3.

@Lippsson don't you already use a barcode scanner successfully?

Lippsson, 2024-05-11:
Yes true, I use this model for scanning. The advantage of this reader is that there is no real laser which can damage kids eyes. Tera Streckkodsskanner 1D 2D QR... https://www.amazon.se/dp/B098DCGK91?ref=ppx_pop_mob_ap_share

s-martin, 2024-05-11:
As this would be more a howto than a feature request, I would convert this issue into a Q&A discussion.

Anybody interested in providing some hints how to use a barcode scanner could add a  comment here.