MiczFlor/RPi-Jukebox-RFID #564: Install script for Raspbian Buster - first steps and help needed!

MiczFlor, 2019-08-18:
Hi everybody, 
Thanks to @ckuetbach and @AdmiralVS I got something to work with when making a new **one-line-install-script** for the latest rapsbian release: buster.

It's currently only on the `develop` branch. How to use this version of the "one line install" see below or go [straight to the wiki](https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/INSTALL-stretch#one-line-install-command).

Help would be welcome to test and improve it. And even more welcome to migrate the new version to the *+Spotify* edition of the Phoniebox.

Things which might soon be *known issues*, at the moment the script *works for me*, though:

* file upload does not work, because the php.ini was not replaced
* static IP address assignment 
* ... add your findings here in the thread

Phoniebox Classic (buster):
~~~
cd; rm buster-install-*; wget https://raw.githubusercontent.com/MiczFlor/RPi-Jukebox-RFID/develop/scripts/installscripts/buster-install-default.sh; chmod +x buster-install-default.sh; ./buster-install-default.sh
~~~

AdmiralVS, 2019-08-23:
The short version: Success! The slightly longer one: I didn't run the installer directly but executed all steps after the software installation one by one and got a box that was working; at least to the point where I can access the web interface and play music from there. I didn't try anything else, yet, and might not get around to doing so within the next weeks.
So: Thanks once more for another great piece of work!

dMopp, 2019-09-27:
Well.. did this today without knowing that one..

Issues i had:

php packages --> Change to 7.3 instead of 7.0
modipy packag source --> changed to buster
lighttp --> config broken, fixed that with default values AND modified php-fpm to use php 7.3 socket

MiczFlor, 2019-09-28:
Hi @dMopp 
did you use the spotify version? I haven't made a spotify version for the install script.

dMopp, 2019-09-28:
Yes,

It’s not complete (l didn’t modify the lighttpd part) but at least the package part

Mit freundlichen Grüßen
Daniel Quolke

deisi, 2019-11-01:
It works, but there are some quirks.
It e.g. sets the ownership of files in `RPi-Jukebox-RFID/` to root. As a result logging wasn't working properly. As a fix i just chowned the complete folder after creation with:
`sudo chown pi:pi RPi-Jukebox-RFID/`

JoeyRamone, 2020-01-05:
First I want so say "Thank you" for all the time you`ve spent for this amazing project. I`ve started some days ago with an raspi 4 and tried the buster image... now I am totally confused and tried to install the spotify version several times without success. 
My understanding was, that the one-line installation script with spotify for buster should work now, right? 
I`ve used raspbian lite for the pi and found no GIT installation by default, so the script fails. after installation of GIT, some more steps working, but still a lot of failures. Before I go on, I would like to ask, if the spotify version is already "ready to go" or still to develop for buster? And if is should be based on raspbian lite, because of the missing git package?

I am new to raspi and programming....

Squelsh, 2020-01-05:
@JoeyRamone 
At the current state there is a problem with incompatible library versions. See https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/688#issuecomment-570145271

I still try to find a workaround by using older versions of Mopidy but until now I had no success.

JoeyRamone, 2020-01-05:
@Squelsh Thanks for the info. I've just opened a new issue, before I saw your comment. Good to know.

mikebarkmin, 2020-01-06:
@MiczFlor I have opend a pull request, which might solve some problems with mopidy3 #701

s-martin, 2020-02-19:
As there's #721 this could be fixed.

closed 2020-02-19 by s-martin