MiczFlor/RPi-Jukebox-RFID #566: WebInterface is displaying wrong play time / overall length

Berdsen, 2019-08-19:
**Description**
When having long playlists above an hour, the webinterface starts displaying the time with an hour already added. This seems to be a problem with daylight saving times and the php function date.

**Technical description**
Anywhere where the date function is used in the php code like `.date("H:i:s", date).` the raspberries(?) current settings will be used to determine the daylight saving time and probably add an hour, if it is currently active.

**Affected files:**
- htdocs/ajax.loadTime.php
- htdocs/ajax.loadOverallTime.php
- htdocs/ajax.loadPlaylist.php

I'll fill up a pull request to use gmdate instead of date for the desired parts.

[image]

[image]

MiczFlor, 2019-08-23:
Hi @Berdsen 
nice catch!

MiczFlor, 2019-08-28:
Hi @Berdsen 
I added your files to the develop branch, could you please take a look and confirm they are working?
All the best, micz

Berdsen, 2019-09-01:
Hi @MiczFlor
I'll check it this week and leave a comment here, when I'm done.
All the best

Berdsen, 2019-09-10:
Hi @MiczFlor,

took a little longer to test. I switched to the latest develop branch on my phoniebox and checked, that it will work perfect :)
This issue can be closed now or when it will be merged to master.

PR #567 (2019-08-23): Using gmdate instead of date
Using gmdate instead of date where format is displaying hour part. Fixes #566
commit e07f5b18 (2019-08-23): Using gmdate instead of date where format is displaying hour part. Fixes #566
closed 2019-09-13 by MiczFlor