MiczFlor/RPi-Jukebox-RFID #570: problems with ampersand "%" in folder name

arnschi, 2019-08-22:
Hi,
Running on classic edition 1.2-rc1 I have a problem with a folder that contains an "&".

Folder "Bob Marley & the Wailers":
Working: I can assign a RFID card via webinterface to this folder
Working: the created shortcut file contains the correct folder name
Not working: swiping the card does not play anything
Not working: clicking the folder in web interface does not play anything
Not working: expanding the folder in web view ("show content") does not work

renaming the folder in "Bob Marley and the Wailers":
Everything works great and as expected

regards,
arnschi

andreasbrett, 2019-10-07:
Hi @arnschi,
I think I could fix all of those issues. Could you please test my code. You can find it in https://github.com/MiczFlor/RPi-Jukebox-RFID/pull/599

MiczFlor, 2019-10-08:
I committed the code to master: #600
thanks for the input!

closed 2019-10-08 by MiczFlor

andreasbrett, 2019-10-08:
Thanks for the quick merge.