MiczFlor/RPi-Jukebox-RFID #572: Pause Audio if card is removed, Resume if swipe again.

jebos, 2019-09-02:
I would like to pause the audio if the card is removed. If it's removed for longer than 1 Minute I would like to stop the playback. If the card is placed again on the reader, the playback should resume or if it was stopped it should restart.

I think this could be done by sending artificial card ids in the reader.py... 

what do you think?

jebos, 2019-09-02:
I did some hacking. I changed Reader.py (using exprimental as base, since I run a RM6300). 
I added a counter to count the number of empty reads (reads without a card), of course I added a else: to the if read_byte == b'\x02':

There I check if the counter is equals 10, and set a card_id "stop". While below 10 I set a card_id "pause" and add one to the counter. 

this causes that the card eval receives a stop or pause card - which i did configure in the rfid_trigger_play.conf.

It works as expected, but it's still a hack. What would be better way to implement such feature?

MiczFlor, 2019-09-05:
Hi @jebos 
could you create a pull request agains the develop branch, so I can take a look at the code? Possibly the changes can be merged even though it's a edge case - and then be documented in the wiki?

jebos, 2019-09-10:
I will check if I find a better way to implement it (less hacky) nevertheless you find my changes in my fork (Feature branch)

jebos, 2019-09-23:
Hey, my Kids actually played around with it and decided they do not like this. We are now at a "Controll Card" Mechanism which is kind of changed to the Master Behavior in a way that they can keep the card on the reader and have dedicated pause  / stop cards, but pause wont toggle the playback. they seem happy, I close this request.

closed 2019-09-23 by jebos