MiczFlor/RPi-Jukebox-RFID #574: Is it possible to skip to the next folder with card or button?

Worren, 2019-09-04:
Is there a possibility to use a rfid card or button to skip through the different folders?
Our childrens now have so many cards, that it would be cool, if they can choose a playback folder without having the right card at hand.

MiczFlor, 2019-09-05:
Hi @Worren 
Interesting thought, I will put this on my list. It is of course a bit like brainless channel hoping or dialing through radio stations along the frequency. But in each cases, well, this is what people do. So it makes sense :)

Worren, 2019-09-05:
Our phoniebox have displays, so there is at least some information.
Unfortunately we forgot some cards in vacation ... and the wifi connection was back at home ...

dnmtch, 2019-09-05:
> and the wifi connection was back at home ...
if only AP-Mode would be possible....

Worren, 2020-02-20:
Any Idea how to hack this?  Next vacation is coming and I am a little bit tired to make rfid cards :-)

s-martin, 2020-02-20:
You would need to adapt `gpio-buttons.py`, `playout_controls.sh` and then find a solution for moc to play next folder, maybe with `load <playlist>`, see also https://linux.die.net/man/1/mpc

You could check how next song is implemented.

Worren, 2020-02-20:
I already found that I could use rfid_trigger_play.sh -d='foldername' to play a folder. And I know how to call the command with an rfid tag. But I don't have a good idea how to cycle through the different folders in a easy way.

rimeraz, 2020-12-19:
@Worren: I have the same requirement. 

And my current idea would be to "glue" in another script into the rid_trigger_play.sh that determines the next folder based on the current playlist. 

I thought of something like in https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Hack-Playlist-triggered-by-GPIO-Button:
```
def def_nextfolder():
    check_call("./scripts/rfid_trigger_play.sh --cardid=$(./scripts/getNextFolder.sh)", shell=True)
def def_prevfolder():
    check_call("./scripts/rfid_trigger_play.sh --cardid=$(./scripts/getPrevFolder.sh)", shell=True)
```

`getPrevFolder.sh` and `getNextFolder.sh` could determine prev/next folder $somehow from `RPi-Jukebox-RFID/shared/latestID.txt` which contains the current card. 

Just an idea. Maybe that's a start. After all this phoniebox is still new to me. My first box looks similar but I had re-written [Marco Wiedermeyer's .NET code](https://blog.mwiedemeyer.de/post/2017/Musikbox-fur-Kind-2) in python.

s-martin, 2024-05-10:
For anyone interested in implementing this, you should target the new V3 (`future3` branches).

V3 uses RPC commands (see [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/future3/develop/documentation/builders/card-database.md)), which could be extended for the desired functionality. There are already commands for prev/next song, this could probably be extended.

devbar, 2026-10-13:
I implemented this for my project and have a proposal in a PR draft

#2719