MiczFlor/RPi-Jukebox-RFID #575: Audio not working

crocadila, 2019-09-09:
Yesterday i tried to install the phoniebox. I choose Raspian buster and the 1 line installer. Audio inside Raspian is working fine (playing youtube etc).

But inside the phoniebox I cant change the volume. The files are playing though.

What am i missing?

MiczFlor, 2019-09-10:
Hi @crocadila 
if I understand this correctly, you can play files in folders via Phoniebox. Does that mean web app or RFID cards? (I hope both :)
You can not change the volume level via Phoniebox. Again: web app or RFID card (or both?)

Please type:
~~~
echo -e status\\nclose | nc -w 1 localhost 6600
~~~
This should show you a lot of settings from the music player daemon `mpd`. What does the volume say?

If you have something playing over Phoniebox, try opening the command line, go to the home directory and type: 
~~~
./RPi-Jukebox-RFID/scripts/playout_controls.sh -c=volumeup
~~~
This should increase the volume level. Does it? If you think so, confirm with the above line.

The opposite should also work, the volume goes up:
~~~
./RPi-Jukebox-RFID/scripts/playout_controls.sh -c=volumeup
~~~

Please share here if the command line controls do something.

jebos, 2019-09-14:
can you try a     sudo sysremctl restart mpd  and check if you can change the volume now?

crocadila, 2019-09-14:
Hello

i got help from a "programmerfriend"  I tried this:

`. RPi-Jukebox-RFID/scripts/playout_controls.sh -c=setvolume -v=80`

and got this response:

```
OK MPD 0.21.4
OK
```

and jebos I tried sudo systemctl resart mpd and nothing changed.

MiczFlor, 2019-09-28:
Hi @crocadila 
In your first post you said "I cant change the volume. The files are playing though." in the latest you say "change the Volume via Webinterface/RFID/GPIO it always stays at 0".
Doest that mean that using Phoniebox:

a) the files play but I hear nothing - only the seconds progress in the web app player
b) the files play and I can hear them, but the volume always shows 0 as volume, trying to change the volume does nothing
c) I can hear the files and the volume changes using the web app, but the display always shows volume 0

Thanks

afiller, 2019-09-28:
I have the same problem with the buster release on the RPI 3 b: 
The startup sound works, but no audio is played/can be heart and the volume control is always on 0. How can I help to solve the issue. I have time this evening for ideas.

afiller, 2019-09-28:
PS: No playback progress is shown, so maybe the playback itself is not working and it is no audio issue.

afiller, 2019-09-28:
PPS: I just looked deeper into the system and the syslog reports the following whenever I want to play file:

```
Sep 28 18:52:58 raspberrypi lighttpd[803]: mpd error: Connection refused
Sep 28 18:52:58 raspberrypi lighttpd[803]: mpd error: Connection refused
Sep 28 18:52:58 raspberrypi lighttpd[803]: mpd error: Connection refused
```

MiczFlor, 2019-09-29:
Hi @afiller did you do the new install with the buster install script?
@crocadila if you have a second SD card around, this might be worth a try? I know the chances are slim that there are two corrupt SD cards creating the same problem. But there is the chance :)

afiller, 2019-09-29:
@MiczFlor Yes, I used the script for the setup. I only used raspi-config to enlarge the partition prior to start.

Fl0ete, 2019-10-01:
Audio isn't working for me either after a fresh install. 
For me it doesn't even say it's playing in the web gui. 
Volume doesn't seem to be 0.

echo -e status\\nclose | nc -w 1 localhost 6600

OK MPD 0.21.4
volume: 100
repeat: 0
random: 0
single: 0
consume: 0
playlist: 4
playlistlength: 0
mixrampdb: 0.000000
state: stop
OK

I already tried another SD card and another Pi (both 3 b+).

MarkIpswich, 2019-10-04:
I am having this problem also.

I change the volume in the web app but after setting it, it goes back to 0%.

What do I need to do to fix this please?

crocadila, 2019-10-07:
Hello again. 
I formated the card again and reinstalled everything.  Last time I couldnt hear anything but i could see the seconds progress.
this time i cant hear anything and the files dont seem to play. I cant change the volume either.

MiczFlor, 2019-10-08:
Hi @Fl0ete 
can you hear anything when playing audiofiles from the command line or gui outside of phoniebox?

@crocadila 
same question to you, see above.

Fl0ete, 2019-10-08:
Here is what worked for me:
https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/592#issuecomment-537567923

crocadila, 2019-10-08:
@MiczFlor No i cant hear anything. The files is playing though (after using @Fl0ete s working solution) I was missing the "playlists" folder too.

s-martin, 2020-03-09:
Is this still an issue?

afiller, 2020-03-09:
Not for me anymore.

s-martin, 2020-03-09:
Closing this issue.

If you still experience audio problems, please check https://github.com/MiczFlor/RPi-Jukebox-RFID/wiki/Troubleshooting-FAQ#audio-is-not-working and/or install a recent version.

If you still have audio problems, please open a new issue.

closed 2020-03-09 by s-martin