MiczFlor/RPi-Jukebox-RFID #576: Buttons do not work (phoniebox does not respond to the incoming signal)

rubynubee, 2019-09-10:
Hi, first of all, thank you guys, for this great project!
You made it very easy to setup the jukebox even for absolute beginners.
Previously, I was able to set up the phoniebox with a pi3B, including a USB soundcard and working RFID control.

But now I stuck the buttons do not want to work. More specifically, the hardware seems to work, but the phoniebox does not respond to the incoming signal.

1. I checked the connection between the hardware pins / GPIO numbers and the set numbers in the gpio-buttons.py script. As well, except that nothing happens, if i hit the button. I even tried the key assignment "flexible layout", it does not work either.

2. To check if my soldered buttons have any effect, I use this GPIO monitoring and testing script (https://www.raspberrypi.org/forums/viewtopic.php?t=167609).
When I run the script and press a button, a value change is displayed. So they seem to work right.

```
`                        * Raspberry Pi GPIO monitor *                         |

| 3 Model B 1Gb detected (26 lines)                    *   Debounce 200 ms     |

|     GPIO2  = True(v)   |      GPIO11  = False(v)   |      GPIO20 = False(v)  |
|     GPIO3  = True(v)   |      GPIO12  = False(v)   |      GPIO21 = False(v)  |
|     GPIO4  = False(v)  |      GPIO13  = False(v)   |      GPIO22 = False(v)  |
|     GPIO5  = False(v)  |      GPIO14  = False(v)   |      GPIO23 = False(v)  |
|     GPIO6  = False(v)  |      GPIO15  = False(v)   |      GPIO24 = False(v)  |
|     GPIO7  = False(v)  |      GPIO16  = False(v)   |      GPIO25 = False(v)  |
|     GPIO8  = False(v)  |      GPIO17  = False(v)   |      GPIO26 = False(v)  |
|     GPIO9  = False(v)  |      GPIO18  = False(v)   |      GPIO27 = True(v)   |
|     GPIO10 = False(v)  |      GPIO19  = False(v)   |                         |

2019-Sep-10 13:26:02: Channel 22 changed (off)
2019-Sep-10 13:26:02: Channel 27 changed (off)
2019-Sep-10 13:26:02: Channel 22 changed (off)
2019-Sep-10 13:26:03: Channel 22 changed (off)
2019-Sep-10 13:26:03: Channel 27 changed (on) `
```

3. I also checkt the status of the gpio-buttons.py, it's displayed as aktiv.

Now I'm out of a clue to fix the problem. Any ideas?

MiczFlor, 2019-09-13:
Hi @rubynubee 
looks like you use USB for soundcard AND RFID? sometimes there is a conflict between sound cards and / or RFID readers which use pins as well - and the gpio daemon of phoniebox.

MiczFlor, 2019-09-28:
Hi @rubynubee 
I was about to suggest that: get it to work first without the usb soundcard, without spotify . Plain vanilla, as they say. Did you get any results?

closed 2020-01-18 by MiczFlor