MiczFlor/RPi-Jukebox-RFID #578: Fast Forward and Rewind via GPIO-Buttons

rpkZarathustra, 2019-09-13:
How about fast forward and rewind like in the web-interface but with GPIO-buttons. In case of large sound files this could be handy.

Or maybe it ist possible that a long or a short push of the SKIP-buttons is read as fast forward/rewind or skip.

iwishitwassummer, 2019-09-26:
Hi @rpkZarathustra  
I did this and it comes in totally handy (couldn't stand the beeping noise of the RFID reader).
I used [those](https://www.ebay.de/itm/173294908693) buttons. You need to solder as described [here](https://www.roboter-bausatz.de/media/pdf/78/84/b7/HTTM-Schalter-Datenblatt.pdf) in section 5.

code should be added to gpio-buttons.py, please check pins and enter yours (I'm actually using splitti's oled script for this, hence the jukebox variable):
`
jukebox4kidsPath = "/home/pi/RPi-Jukebox-
def def_seekahead():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playerseek -v=+20", shell=True)

def def_seekback():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playerseek -v=-20", shell=True)
seekA = Button(16,pull_up=False,hold_time=0.3,hold_repeat=True)
seekB = Button(19,pull_up=False,hold_time=0.3,hold_repeat=True)

seekA.when_pressed = def_seekahead
seekA.when_held = def_seekahead
seekB.when_pressed = def_seekback
seekB.when_held = def_seekback
`
This works as I expected, short press +/- 20 sec and if hold is seeks ahead/backwards as long as it's hold

(see also [here](https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/450))

MiczFlor, 2019-09-28:
Hi @iwishitwassummer 
thumbs up. Could you create a pull request against the `develop` branch?

iwishitwassummer, 2019-09-29:
@MiczFlor I'll try, but I'm a newbie to working with code in github

MiczFlor, 2019-09-29:
Alternatively, you could post the complete file here in the thread and I will merge it in.

carver79, 2019-10-17:
Could you post your gpio-buttons.py?
Eversince i try to "merge" your code with mine (also Splitti), i cant activate the GPIO-Buttons. Just to compare and search the bug in my copy-paste-coding.

iwishitwassummer, 2019-10-18:
Apologies for the late reply
Is placed in ~/oled_phoniebox/scripts/gpio-buttons

```
#!/usr/bin/python3
from gpiozero import Button`
from signal import pause
from subprocess import check_call
from time import sleep
# This script will block any I2S DAC e.g. from Hifiberry, Justboom, ES9023, PCM5102A
# due to the assignment of GPIO 19 and 21 to a buttons

# 2018-10-31
# Added the function on holding volume + - buttons to change the volume in 0.3s interval
#
# 2018-10-15
# this script has the `pull_up=True` for all pins. See the following link for additional info:
# https://github.com/MiczFlor/RPi-Jukebox-RFID/issues/259#issuecomment-430007446
#
# 2017-12-12
# This script was copied from the following RPi forum post:
# https://forum-raspberrypi.de/forum/thread/13144-projekt-jukebox4kids-jukebox-fuer-kinder/?postID=312257#post312257
# I have not yet had the time to test is, so I placed it in the misc folder.
# If anybody has ideas or tests or experience regarding this solution, please create pull requests or contact me.

jukebox4kidsPath = "/home/pi/RPi-Jukebox-RFID"

def def_shutdown():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=shutdown", shell=True)

def def_volU():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=volumeup", shell=True)

def def_volD():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=volumedown", shell=True)

def def_vol0():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=mute", shell=True)

def def_next():
#  for x in range(0, 19):
#    if next.is_pressed == True :
#      sleep(0.1)
#    else:
      check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playernext", shell=True)
#      break

def def_contrastup():
  if prev.is_pressed == True :
    check_call("/usr/bin/touch /tmp/o4p_overview.temp", shell=True)
  else:
    check_call("/usr/bin/python3 /home/pi/oled_phoniebox/scripts/contrast/contrast_up.py", shell=True)

def def_contrastdown():
  if next.is_pressed == True :
    check_call("/usr/bin/touch /tmp/o4p_overview.temp", shell=True)
  else:
    check_call("/usr/bin/python3 /home/pi/oled_phoniebox/scripts/contrast/contrast_down.py", shell=True)

def def_prev():
#  for x in range(0, 19):
#    if prev.is_pressed == True :
#      sleep(0.1)
#    else:
      check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playerprev", shell=True)
#      break

def def_halt():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playerpause", shell=True)

def def_seekahead():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playerseek -v=+20", shell=True)

def def_seekback():
    check_call(jukebox4kidsPath+"/scripts/playout_controls.sh -c=playerseek -v=-20", shell=True)

#shut = Button(3, hold_time=2)
#vol0 = Button(13,pull_up=True)
#volU = Button(16,pull_up=True,hold_time=0.3,hold_repeat=True)
#volD = Button(19,pull_up=True,hold_time=0.3,hold_repeat=True)
next = Button(26,pull_up=False,hold_time=0.1,hold_repeat=False)
prev = Button(20,pull_up=False,hold_time=0.1,hold_repeat=False)
#prev = Button(20,pull_up=False)
#next = Button(26,pull_up=False)
halt = Button(21,pull_up=False)
# vorspulen/zurueckspulen ACHTUNG sind dieselben pins wie Volume im classic mode der Phoniebox
seekA = Button(16,pull_up=False,hold_time=0.3,hold_repeat=True)
seekB = Button(19,pull_up=False,hold_time=0.3,hold_repeat=True)

#shut.when_held = def_shutdown
#vol0.when_pressed = def_vol0
#volU.when_pressed = def_volU
#When the Volume Up button was held for more than 0.3 seconds every 0.3 seconds he will call a ra$
#volU.when_held = def_volU
#volD.when_pressed = def_volD
#When the Volume Down button was held for more than 0.3 seconds every 0.3 seconds he will lower t$
#volD.when_held = def_volD
seekA.when_pressed = def_seekahead
seekA.when_held = def_seekahead
seekB.when_pressed = def_seekback
seekB.when_held = def_seekback
next.when_pressed = def_next
#next.when_held = def_contrastup
prev.when_pressed = def_prev
#prev.when_held = def_contrastdown
halt.when_pressed = def_halt

pause()
```

carver79, 2019-10-19:
Thx - i dont have any coding skills, but every volume is commented out. 
how do you change volume?

iwishitwassummer, 2019-10-19:
Remove # at beginning of the line. 
Be sure the pins are the ones you use

donhomie, 2019-11-28:
Thanks for the code, just managed to get it to work! In case anyone is having trouble getting this to work: try using pull_up=True, so e.g.:

seekA = Button(16,pull_up=False,hold_time=0.3,hold_repeat=True)

thedipling, 2020-03-16:
Hello There, is there any progress or implementation to fastforward with the next button and reverse/previous comming? this would be wonderful!

Since I am not able to code my self... i would appreciate your help :)

cheers ! and thanks for your hard Work

iwishitwassummer, 2020-03-25:
@thedipling just try coding, it's fun :)
and you might need to adapt the script most likely to your GPIO pins anyhow.
in your phoniebox folder go to the _script_ folder and open _gpio-buttons.py_ in a coding editor

Look at my post from October on where to those follwing lines.
Best those under all the other line with def at the beginning
`def def_seekahead():
    check_call("./scripts/playout_controls.sh -c=playerseek -v=+20", shell=True)`
`def def_seekback():
    check_call("./scripts/playout_controls.sh -c=playerseek -v=-20", shell=True)`

then those somewhere in the middle:
`seekA = Button(16,pull_up=False,hold_time=0.3,hold_repeat=True)`
`seekB = Button(19,pull_up=False,hold_time=0.3,hold_repeat=True)`

somewhere in the end:
`seekA.when_pressed = def_seekahead`
`seekA.when_held = def_seekahead`
`seekB.when_pressed = def_seekback`
`seekB.when_held = def_seekback`

the tricky bit in python is the indentation. If you use a good coding editor it will help to show you "the empty spaces", try to align it to the "paragraph" your in e.g. align with the oder _def_

thedipling, 2020-03-25:
@iwishitwassummer thanks! I've red it - and already tried to implement it - but... i was not successful.. I thought there is something missing...

but i will try it again :)

btw. my editor is nano ^^ 

another silly question - the pins 16 and 19 are the next and previous pins in your settings?

and there is nothing else to do? What is the time which will be skiped?

That would be greate if that works

iwishitwassummer, 2020-03-25:
my next and prev buttons are 20 and 26:
`next = Button(26,pull_up=False,hold_time=0.1,hold_repeat=False)`
`prev = Button(20,pull_up=False,hold_time=0.1,hold_repeat=False)`

 16 and 19  are used for seeking forth and back. But be aware in Phoniebox Classic those are used for Volume settings so you most likely have to adapt the pins to match yours

time skipped is defined in the def with parameter -v that a mpd parameter in sec:
`def def_seekback(): check_call("./scripts/playout_controls.sh -c=playerseek -v=-20", shell=True)`
Minus or Plus for backwards or forwards seeking

the def reference the script [playout_control](https://github.com/MiczFlor/RPi-Jukebox-RFID/blob/53ecb53f3363f18e627e282a329bc27f6a4622f7/scripts/playout_controls.sh#L360) where everything was already defined

Just FYI: I'm still on 1.1.9 Classic, I am trying to update but that involves the new OS, hence bigger job, but I don't think it has changed

thedipling, 2020-03-25:
ah okay I am on 2.0 rc8 (I think)

here is my config attached.

[gpio-buttons_RPi1B_thedipling.py.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4383772/gpio-buttons_RPi1B_thedipling.py.txt)

I hope you or someone else could help :)

iwishitwassummer, 2020-03-26:
Don't see a difference. also no change in playout controls as far as I can see
But do you have a version where you added my code?

thedipling, 2020-03-26:
thats my working config. I've added yours but it was not working so i decided to upload the working one - if it is possible would you add your code so that it should work?

otherwise I try to upload the not-working config tomorrow evening.

cheers and thanks for your replies so far

thedipling, 2020-03-30:
Sorry for the late answer.
here you can find the modified - not working py-file. 

with the previous one the cpu is normal at ca. 20% - with the modified one at 100% and the buttons are not working. 

cheers
[gpio-buttons-seek-test_notsoworking_RPi1B.py.txt](https://github.com/MiczFlor/RPi-Jukebox-RFID/files/4404871/gpio-buttons-seek-test_notsoworking_RPi1B.py.txt)

iwishitwassummer, 2020-03-30:
your next/back are on the same pin. I could that to work as well, there are not enough status for a button I guess. Use different pins

Caliandroid, 2021-02-07:
One simple way to achieve Fast Forward/Rewind/Seek via Buttons
Add the following line to the gpio_settings.ini in the settingsfolder

[FastForward]
enabled: True
Type:  Button
Pin: YOUR GPIO PIN
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerSeekFwd

[Rewind]
enabled: True
Type:  Button
Pin: YOUR GPIO PIN
pull_up: True
hold_time: 0.3
functionCall: functionCallPlayerSeekBack

@MiczFlor  Merge request with this lines created

closed 2021-02-15 by s-martin