MiczFlor/RPi-Jukebox-RFID #579: playlist playback loops even it is not activated

d3rmadd1n, 2019-09-22:
Hi,

My jukebox is always looping the playlist, even it is not activated. Is there a global setting which may overrides the individual setting for each playlist? 

Thanks :-)

MiczFlor, 2019-09-28:
Hi @d3rmadd1n 
hm. Shouldn't be. I will look into it. Fun fact: a year ago when we migrated from VLC to mpd the fact that the playlist does "not" loop was a problem to some :)

d3rmadd1n, 2019-09-28:
Hi MiczFlor,

Thank you for your reply. I have just reinstalled the whole jukebox yesterday and it worked again as it should be. I am sorry for the late reply. 

Martin

closed 2019-09-28 by d3rmadd1n

sassoz, 2020-04-27:
Hi @MiczFlor,
I have the same issue as above. I've tried reinstalling, but still, if I set the loop to off in the web gui, it is set always as loop on per default. Can I change this behaviour somewhere?
Thanks
Angelo

sassoz, 2020-05-02:
hi, 
I have the same issue. The loop setting is set on by default and each card/playlist always play in loop mode. Even after setting loop off, it will revert to loop on. The settings in the folder.conf is still loop=OFF